llvm/circt #1834: Python binding on aarch64 fails

oroppas, 2021-09-22:
```
uname -a
Linux raspi 5.10.63-1-MANJARO-ARM #1 SMP PREEMPT Wed Sep 8 14:17:41 UTC 2021 aarch64 GNU/Linux
```

Here's an example

```
********************
FAIL: CIRCT :: Bindings/Python/dialects/hw.py (2 of 28)
******************** TEST 'CIRCT :: Bindings/Python/dialects/hw.py' FAILED ********************
Script:
--
: 'RUN: at line 2';   /usr/bin/python3.9 /home/ryuta/packages/circt/circt/integration_test/Bindings/Python/dialects/hw.py | /home/ryuta/packages/circt/build/bin/FileCheck /home/ryuta/packages/circt/circt/integration_test/Bindings/Python/dialects/hw.py
--
Exit Code: 2

Command Output (stderr):
--
 #0 0x0000ffff956d32a0 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) /home/ryuta/packages/circt/circt/llvm/llvm/lib/Support/Unix/Signals.inc:565:22
 #1 0x0000ffff956d3370 PrintStackTraceSignalHandler(void*) /home/ryuta/packages/circt/circt/llvm/llvm/lib/Support/Unix/Signals.inc:632:1
 #2 0x0000ffff956d1208 llvm::sys::RunSignalHandlers() /home/ryuta/packages/circt/circt/llvm/llvm/lib/Support/Signals.cpp:97:20
 #3 0x0000ffff956d2b38 SignalHandler(int) /home/ryuta/packages/circt/circt/llvm/llvm/lib/Support/Unix/Signals.inc:407:1
 #4 0x0000ffff967be7c0 (linux-vdso.so.1+0x7c0)
 #5 0x0000ffff912047ac mlir::ArrayAttr::getValue() const /home/ryuta/packages/circt/build/tools/mlir/include/mlir/IR/BuiltinAttributes.cpp.inc:105:77
 #6 0x0000ffff90ff965c mlir::ArrayAttr::size() const /home/ryuta/packages/circt/build/tools/mlir/include/mlir/IR/BuiltinAttributes.h.inc:70:43
 #7 0x0000ffff91d93c04 mlir::ArrayAttr::empty() const /home/ryuta/packages/circt/build/tools/mlir/include/mlir/IR/BuiltinAttributes.h.inc:71:43
 #8 0x0000ffff92a5e63c printParameterList(mlir::ArrayAttr, mlir::OpAsmPrinter&) /home/ryuta/packages/circt/circt/lib/Dialect/HW/HWOps.cpp:614:23
 #9 0x0000ffff92a5e7d0 printModuleOp(mlir::OpAsmPrinter&, mlir::Operation*, ExternModKind) /home/ryuta/packages/circt/circt/lib/Dialect/HW/HWOps.cpp:648:8
#10 0x0000ffff92a5e9a0 printHWModuleOp(mlir::OpAsmPrinter&, circt::hw::HWModuleOp) /home/ryuta/packages/circt/circt/lib/Dialect/HW/HWOps.cpp:675:29
#11 0x0000ffff92a702f4 circt::hw::HWModuleOp::print(mlir::OpAsmPrinter&) /home/ryuta/packages/circt/build/tools/circt/include/circt/Dialect/HW/HW.cpp.inc:2111:36
#12 0x0000ffff92a49930 mlir::Op<circt::hw::HWModuleOp, mlir::OpTrait::OneRegion, mlir::OpTrait::ZeroResult, mlir::OpTrait::ZeroSuccessor, mlir::OpTrait::ZeroOperands, mlir::OpTrait::IsIsolatedFromAbove, mlir::OpTrait::FunctionLike, mlir::SymbolOpInterface::Trait, mlir::RegionKindInterface::Trait, mlir::OpTrait::SingleBlockImplicitTerminator<circt::hw::OutputOp>::Impl, mlir::OpTrait::HasParent<mlir::ModuleOp>::Impl>::printAssembly(mlir::Operation*, mlir::OpAsmPrinter&, llvm::StringRef) /home/ryuta/packages/circt/circt/llvm/mlir/include/mlir/IR/OpDefinition.h:1796:42
#13 0x0000ffff91247354 void llvm::detail::UniqueFunctionBase<void, mlir::Operation*, mlir::OpAsmPrinter&, llvm::StringRef>::CallImpl<void (* const)(mlir::Operation*, mlir::OpAsmPrinter&, llvm::StringRef)>(void*, mlir::Operation*, mlir::OpAsmPrinter&, llvm::StringRef) /home/ryuta/packages/circt/circt/llvm/llvm/include/llvm/ADT/FunctionExtras.h:216:49
#14 0x0000ffff911cbefc llvm::unique_function<void (mlir::Operation*, mlir::OpAsmPrinter&, llvm::StringRef) const>::operator()(mlir::Operation*, mlir::OpAsmPrinter&, llvm::StringRef) const /home/ryuta/packages/circt/circt/llvm/llvm/include/llvm/ADT/FunctionExtras.h:404:62
#15 0x0000ffff911ca0b0 mlir::AbstractOperation::printAssembly(mlir::Operation*, mlir::OpAsmPrinter&, llvm::StringRef) const /home/ryuta/packages/circt/circt/llvm/mlir/include/mlir/IR/OperationSupport.h:102:49
#16 0x0000ffff911b548c (anonymous namespace)::DummyAliasOperationPrinter::print(mlir::Operation*) /home/ryuta/packages/circt/circt/llvm/mlir/lib/IR/AsmPrinter.cpp:376:9
#17 0x0000ffff911b593c (anonymous namespace)::DummyAliasOperationPrinter::print(mlir::Block*, bool, bool) /home/ryuta/packages/circt/circt/llvm/mlir/lib/IR/AsmPrinter.cpp:431:26
#18 0x0000ffff911b59dc (anonymous namespace)::DummyAliasOperationPrinter::printRegion(mlir::Region&, bool, bool, bool) /home/ryuta/packages/circt/circt/llvm/mlir/lib/IR/AsmPrinter.cpp:444:47
#19 0x0000ffff912340c0 mlir::ModuleOp::print(mlir::OpAsmPrinter&) /home/ryuta/packages/circt/build/tools/mlir/include/mlir/IR/BuiltinOps.cpp.inc:476:1
#20 0x0000ffff9124308c mlir::Op<mlir::ModuleOp, mlir::OpTrait::OneRegion, mlir::OpTrait::ZeroResult, mlir::OpTrait::ZeroSuccessor, mlir::OpTrait::ZeroOperands, mlir::OpTrait::AffineScope, mlir::OpTrait::IsIsolatedFromAbove, mlir::OpTrait::NoRegionArguments, mlir::OpTrait::SymbolTable, mlir::SymbolOpInterface::Trait, mlir::OpAsmOpInterface::Trait, mlir::OpTrait::NoTerminator, mlir::OpTrait::SingleBlock, mlir::RegionKindInterface::Trait, mlir::OpTrait::HasOnlyGraphRegion>::printAssembly(mlir::Operation*, mlir::OpAsmPrinter&, llvm::StringRef) /home/ryuta/packages/circt/circt/llvm/mlir/include/mlir/IR/OpDefinition.h:1796:42
#21 0x0000ffff91247354 void llvm::detail::UniqueFunctionBase<void, mlir::Operation*, mlir::OpAsmPrinter&, llvm::StringRef>::CallImpl<void (* const)(mlir::Operation*, mlir::OpAsmPrinter&, llvm::StringRef)>(void*, mlir::Operation*, mlir::OpAsmPrinter&, llvm::StringRef) /home/ryuta/packages/circt/circt/llvm/llvm/include/llvm/ADT/FunctionExtras.h:216:49
#22 0x0000ffff911cbefc llvm::unique_function<void (mlir::Operation*, mlir::OpAsmPrinter&, llvm::StringRef) const>::operator()(mlir::Operation*, mlir::OpAsmPrinter&, llvm::StringRef) const /home/ryuta/packages/circt/circt/llvm/llvm/include/llvm/ADT/FunctionExtras.h:404:62
#23 0x0000ffff911ca0b0 mlir::AbstractOperation::printAssembly(mlir::Operation*, mlir::OpAsmPrinter&, llvm::StringRef) const /home/ryuta/packages/circt/circt/llvm/mlir/include/mlir/IR/OperationSupport.h:102:49
#24 0x0000ffff911b548c (anonymous namespace)::DummyAliasOperationPrinter::print(mlir::Operation*) /home/ryuta/packages/circt/circt/llvm/mlir/lib/IR/AsmPrinter.cpp:376:9
#25 0x0000ffff911b62bc (anonymous namespace)::AliasInitializer::initialize(mlir::Operation*, mlir::OpPrintingFlags const&, llvm::MapVector<mlir::Attribute, (anonymous namespace)::SymbolAlias, llvm::DenseMap<mlir::Attribute, unsigned int, llvm::DenseMapInfo<mlir::Attribute>, llvm::detail::DenseMapPair<mlir::Attribute, unsigned int> >, std::vector<std::pair<mlir::Attribute, (anonymous namespace)::SymbolAlias>, std::allocator<std::pair<mlir::Attribute, (anonymous namespace)::SymbolAlias> > > >&, llvm::MapVector<mlir::Type, (anonymous namespace)::SymbolAlias, llvm::DenseMap<mlir::Type, unsigned int, llvm::DenseMapInfo<mlir::Type>, llvm::detail::DenseMapPair<mlir::Type, unsigned int> >, std::vector<std::pair<mlir::Type, (anonymous namespace)::SymbolAlias>, std::allocator<std::pair<mlir::Type, (anonymous namespace)::SymbolAlias> > > >&) /home/ryuta/packages/circt/circt/llvm/mlir/lib/IR/AsmPrinter.cpp:611:21
#26 0x0000ffff911b679c (anonymous namespace)::AliasState::initialize(mlir::Operation*, mlir::OpPrintingFlags const&, mlir::DialectInterfaceCollection<mlir::OpAsmDialectInterface>&) /home/ryuta/packages/circt/circt/llvm/mlir/lib/IR/AsmPrinter.cpp:731:58
#27 0x0000ffff911cb944 mlir::detail::AsmStateImpl::initializeAliases(mlir::Operation*) /home/ryuta/packages/circt/circt/llvm/mlir/lib/IR/AsmPrinter.cpp:1159:3
#28 0x0000ffff911bf67c mlir::Operation::print(llvm::raw_ostream&, mlir::OpPrintingFlags const&) /home/ryuta/packages/circt/circt/llvm/mlir/lib/IR/AsmPrinter.cpp:2811:10
#29 0x0000ffff9100be04 mlirOperationPrint /home/ryuta/packages/circt/circt/llvm/mlir/lib/CAPI/IR/IR.cpp:418:52
#30 0x0000ffff8ffcb138 mlir::python::populateIRCore(pybind11::module_&)::'lambda27'(mlir::python::PyModule&)::operator()(mlir::python::PyModule&) const /home/ryuta/packages/circt/circt/llvm/mlir/lib/Bindings/Python/IRCore.cpp:2021:36
#31 0x0000ffff90003d44 pybind11::str pybind11::detail::argument_loader<mlir::python::PyModule&>::call_impl<pybind11::str, mlir::python::populateIRCore(pybind11::module_&)::'lambda27'(mlir::python::PyModule&)&, 0ul, pybind11::detail::void_type>(mlir::python::populateIRCore(pybind11::module_&)::'lambda27'(mlir::python::PyModule&)&, std::integer_sequence<unsigned long, 0ul>, pybind11::detail::void_type&&) && /usr/lib/python3.9/site-packages/pybind11/include/pybind11/cast.h:1194:5
#32 0x0000ffff8fffaa4c _ZNO8pybind116detail15argument_loaderIJRN4mlir6python8PyModuleEEE4callINS_3strENS0_9void_typeERZNS3_14populateIRCoreERNS_7module_EEUlS5_E27_EENSt9enable_ifIXntsrSt7is_voidIT_E5value [... cut ...]
      -DCMAKE_BUILD_TYPE=Debug \
      -DLLVM_ENABLE_PROJECTS=mlir \
      -DLLVM_ENABLE_ASSERTIONS=ON \
      -DLLVM_EXTERNAL_PROJECTS=circt \
      -DLLVM_EXTERNAL_CIRCT_SOURCE_DIR=./circt \
      -DMLIR_ENABLE_BINDINGS_PYTHON=ON \
      -DCIRCT_BINDINGS_PYTHON_ENABLED=ON \
      -DCIRCT_BINDINGS_TCL_ENABLED=ON \
      -DCIRCT_ENABLE_FRONTENDS=ON \
      -DVERILATOR_PATH=/usr/bin/verilator

cmake --build build -- -j 1
cmake --build build -- -j 1 check-circt
cmake --build build -- -j 1 check-circt-integration
```

mikeurbach, 2021-09-22:
Hi @oroppas, is it possible this is not aarch64 specific? We currently have an open issue that the Python bindings tests are failing: https://github.com/llvm/circt/issues/1810. I see the same on x86, so hopefully this will be fixed when #1810 is fixed.

oroppas, 2021-09-22:
Hi @mikeurbach. You're absolutely right. After carefully inspecting my x86 build/test, I realized x86 tests were not properly executed and gave me wrong impression. My apologies.

I keep an eye on #1810 and test python binding once it's resolved.

closed 2021-09-22 by oroppas