llvm/circt #1842: [HW Parameters] Improve .mlir syntax

lattner, 2021-09-22:
We're currently using a simple approach to printing and parsing parameter attributes, which means that we get things like this on an instance:

```
hw.module @UseParameters<xxx: i42>() {
  hw.instance "verbatimparam" @NoArg<param: i42 = #hw.param.binary<add #hw.param.verbatim<"xxx">, 17>>() -> () 
}
```

These parameters are only printed in specific contexts (right now: instances, localparams; in the future, in a `!hw.int<param>` sort of type), so we can do much better.  We should be able to get something like:

```
hw.module @UseParameters<xxx: i42>() {
  hw.instance "verbatimparam" @NoArg<param: i42 = xxx + 17>() -> () 
}
```

It may also be worth introducing a consistent sigil to make these things stand out more, and avoid potential keyword ambiguity.  SIL uses the τFoo prefix since it just deals with type parameters.  I think that using a backtick or something that is otherwise unused would make sense.  We would consistently print it in the module parameter list as well as at use-site, e.g. something like:
 
```
hw.module @UseParameters<`xxx: i42>() {
  hw.instance "verbatimparam" @NoArg<`param: i42 = `xxx + 17>() -> () 
```

Maybe ` isn't the best though.

Ramlakshmi3733, 2021-09-29:
On the discussion of superset parameters, I was not thinking of CHISEL or FIRRTL dialect in particular.
Can a front end (any language) emit a set of parameters where:
- some parameters control the transforms in IR passes AND ARE NOT passed into Verilog
- other parameters are passed into Verilog.
Today we have the notion of annotations but these annotations are fixed entities.

lattner, 2021-09-29:
We should probably take this to a different place, this issue is about a particular syntax improvement to parameters in MLIR.    Quick answers: not yet, but we can define that given the right use-cases.