llvm/circt #1848: [HW] Parameters step #4: Add support for "integer" verilog type

lattner, 2021-09-22:
SystemVerilog supports typed integer parameters (e.g. 3 bits) but Verilog doesn't.  We need to support plain old integer types for old verilog clients, this is also the most commonly used thing for type sizes etc.

lattner, 2021-10-04:
Done in 7cdad90d

closed 2021-10-04 by lattner

seldridge, 2021-10-04:
This starts to bump up against some weird edges of Verilog and SystemVerilog around parameters. I'm documenting it here after some experiments with `verilator --lint-only --lanuage 1364-2001` and `verilator --lint-only --language 1800-2017`.

1. Verilog requires that all parameters have a default value.
2. SystemVerilog requires that all parameters are provided a value.

By (1), the following is an illegal circuit:
```verilog
module Foo
  #(parameter [41:0] p1) ();
endmodule

module Top();
  Foo #(.p1(42'b0)) foo;
endmodule
```

```
# verilator -lint-only Tmp.sv --language 1364-2001                 
%Error: Tmp.sv:2:22: Parameter requires default value, or use IEEE 1800-2009 or later.
    2 |   #(parameter [41:0] p1) ();
      |                      ^~
%Error: Exiting due to 1 error(s)
```

SystemVerilog is fine with this, but it is not fine if `module Foo` is standalone as:

```verilog
module Foo
  #(parameter [41:0] p1) ();
endmodule
```

```
# verilator -lint-only Tmp.sv --language 1800-2017
%Error: Tmp.sv:2:22: Parameter without initial value is never given value (IEEE 1800-2017 6.20.1): 'p1'
                   : ... In instance Foo
    2 |   #(parameter [41:0] p1) ();
      |                      ^~
%Error: Exiting due to 1 error(s)
```

What I've usually seen here is that Verilog developers will give a default parameter for everything. This then creates confusion because users of an IP don't know from the code what is a mandatory parameter vs. a parameter with an actual default value (unless comments are included). For CIRCT it makes Verilog generation a little weird as we need to choose a default value for all parameters.

lattner, 2021-10-05:
Thanks @seldridge I think this just means that the frontend needs to provide default values.  I captured this in the rationale here, thanks!  92d6f785