llvm/circt #1857: [HW] Use default parameter

teqdruid, 2021-09-23:
Maybe I'm not doing this correctly, but when I change[ this line ](https://github.com/llvm/circt/blob/d75874d194c1d5f1b731109b2a241d69b15ea4f4/test/Dialect/HW/modules.mlir#L74) to

```mlir
%r2 = hw.instance "inst3" @parameters<>(arg0: %a: i8) -> (out: i8)
```

I get:

```
/home/jodemme/circt/test/Dialect/HW/modules.mlir:74:9: error: 'hw.instance' op expected 2 parameters but had 0
  %r2 = hw.instance "inst3" @parameters<>(arg0: %a: i8) -> (out: i8)
        ^
/home/jodemme/circt/test/Dialect/HW/modules.mlir:74:9: note: see current operation: %2 = "hw.instance"(%arg0) {argNames = ["arg0"], instanceName = "inst3", moduleName = @parameters, parameters = [], resultNames = ["out"]} : (i8) -> i8
/home/jodemme/circt/test/Dialect/HW/modules.mlir:61:1: note: module declared here
hw.module @parameters<p1: i42 = 17, p2: i1>(%arg0: i8) -> (out: i8) {
```

Which implies that there's no way to use default parameter values...? Or is it expected that one inspect the module param decls for the defaults before constructing the instanceop?

lattner, 2021-09-24:
Right, you need to specify all the arguments at the instance site, the default arguments just causes the verilog emitter to omit them.  I'll write up some rationale docs explaining the "why" behind this!

lattner, 2021-09-26:
Done in 039324b8.  PTAL and let me know if this addresses your concern or not, I'd also love feedback and improvements to the rationale docs in general.

closed 2021-09-26 by lattner