llvm/circt #1860: [Calyx][SCFToCalyx] Move control simplification patterns to separate pass

mortbopet, 2021-09-23:
The SCToCalyx pass introduced a variety of simplification and optimization patterns, https://github.com/llvm/circt/pull/1849. The simplification patterns are necessary for allowing the generated Calyx program to pass the Calyx verifiers. However, the simplification patterns can be seen as generally applicable and thus are distinct from the conversion pass itself.

As such, the simplification patterns should be moved to a separate pass which may be run on any Calyx program.

mikeurbach, 2021-09-23:
Sounds great to me. I'd suggest an API along the lines of `populateCalyxSimplificationPatterns(RewritePatternSet &patterns)` for other passes to easily plug in. Another option: do these meet the criteria for [canonicalization](https://mlir.llvm.org/docs/Canonicalization/)? We could define a cost model (probably in a rationale doc), and add patterns that always reduce the cost as canonicalizations.

mortbopet, 2021-09-23:
As far as i can tell, the patterns should fit within the definition of canonicalizations, so that is most likely the way to go!

cgyurgyik, 2021-09-26:
Going to knock this out.

cgyurgyik, 2021-09-27:
For `IfOp`, I'll address this in #1861.

closed 2021-09-27 by cgyurgyik