llvm/circt #1861: [SCFToCalyx] Add simplification pattern for shared parallel groups in if statements

mortbopet, 2021-09-23:
https://github.com/llvm/circt/pull/1849 introduced the `CommonIfTailEnablePattern` for simplifying `if` operations with identical groups in the tail position of each branch's `seq` operation.

A similar simplification pattern can be implemented for shared parallel group statements - any shared `group.enable` within each of the par blocks in the branches can be factored out:
```
if %a with %A {
  par {
    calyx.enable @A
    calyx.enable @B
    calyx.enable @C
  }
} else {
  par {
    calyx.enable @B
    calyx.enable @C
    calyx.enable @D
  }
}
```

simplifies to
```
if %a with %A {
  par {
    calyx.enable @A
  }
} else {
  par {
    calyx.enable @D
  }
}
par {
  calyx.enable @B
  calyx.enable @C
}
```

cgyurgyik, 2021-09-27:
Going to take this on as well. This will require also looking at the `{SeqOp, ParOp}` case.

cgyurgyik, 2021-09-27:
One thing that makes this a bit trickier is the extra information regarding what the `IfOp` is nested in.

PR #1895 (2021-09-30): [Calyx] Add enhancements to common tail elimination for `IfOp`.
This adds a pattern for when `then` and `else` regions are both `ParOp`s. It also takes into consideration the parent region of the `IfOp`, since we don't want to unintentionally change the behavior when pulling out `EnableOp`s. 

I also do not touch anything that's not immediately within the given `ParOp`'s body; doing so may also change the behavior of the program.

The case where the `then` and `else` regions are different, e.g. `SeqOp` and  `ParOp`, should never pull out an `EnableOp`; it will always produce worse code, e.g.
` [... cut]
closed 2021-09-30 by cgyurgyik