llvm/circt #1871: [SCFToCalyx] Add support for `RegionBranchOpInterface`

mortbopet, 2021-09-23:
Currently, `BranchOpInterface` is supported for conversion, which takes care of block-based control flow. 
https://github.com/llvm/circt/blob/533cbc673068f2c4ce668ba63ac78d740246450d/lib/Conversion/SCFToCalyx/SCFToCalyx.cpp#L821-L822

However, operations such as `scf::IfOp` implement region-based control flow. To support this in a generalized manner, an op builder specialization  should be added for the `RegionBranchOpInterface`.

mortbopet, 2021-09-24:
After thinking a bit about this, i think the easiest solution would be to follow the approach of how `scf.execute_region` is handled; inline/lower `RegionOps` to regular control-flow, which can then be handled by the existing `BranchOpInterface` support.

mikeurbach, 2021-09-27:
To me the question is whether we gain something from working at the level of RegionBranchOpInterface. Do we lose information that would be useful to the Calyx conversion? Maybe not, in which case lowering to a CDFG first might make sense. 

I'm looking at a similar question with the Scheduling infrastructure. Can we schedule at the level of RegionBranchOpInterface, and is it preferable to scheduling at the level of a CDFG? I'm not sure yet, but I'll share what I find from that experience.

mortbopet, 2021-09-28:
@mikeurbach so far i'd say no, but that is also because i'm looking at it purely from the context of `scf::IfOp`. Would be good to find some other use-cases which may tell a bit more of the story...

mortbopet, 2021-09-28:
Another thing is that if we do go down the route of lowering to a CDFG, then it would be good to look at how to activate a subset of the conversions in the `SCFToStandard` pass (https://github.com/llvm/llvm-project/blob/47cc166bc023b497bdffe0964d80f15eaee8b7da/mlir/lib/Conversion/SCFToStandard/SCFToStandard.cpp). Currently, what we do to lower `scf::ExecuteRegionOp` is essentially the same as what is done there, and it will be the same for `scf::IfOp`.

mikeurbach, 2021-09-28:
Yep agreed on both points.