llvm/circt #1873: [Calyx] Switch IR form to match native compiler for guards?

cgyurgyik, 2021-09-23:
https://github.com/llvm/circt/blob/069faf81d8535556bb0dcd38b195017936976668/include/circt/Dialect/Calyx/CalyxStructure.td#L291-L292

As mentioned in the TODO, we currently do
```mlir
calyx.assign %dest = %src, %guard ? : i8

calyx.group_done %src, %guard ? : i1 

%result = calyx.group_go %src, %guard ? : i1
```

Something closer to the native compiler would be:
```mlir
calyx.assign %dest = %guard ? %src : i8

calyx.group_done %guard ? %src : i1

%result = calyx.group_go %guard ? %src : i1
```

---

Pros:
- Closer to native compiler form. Could potentially cause confusion otherwise?

Cons:
- Lots of refactoring.
- Need to use a custom parser since the guard is optional.
- Builders need to be reconfigured since the guard is optional.

---

*My thoughts*: IMO, not worth the effort. However, wanted to confirm with you all before dropping this TODO.

mikeurbach, 2021-09-23:
> Need to use a custom parser since the guard is optional.

Does the `assemblyFormat` language not support optional Values in the middle of the op's syntax? I've used optionals before, and you have to anchor the assembly format with a `^` on the optional value, but perhaps this use-case is unsupported.

In general, I don't feel strongly one way or the other. If enough people have their hands in the CIRCT compiler and the Rust compiler, and are hand writing Calyx IR in CIRCT, it might make sense to unify the printed form. Otherwise, I don't know if the change is merited.

cgyurgyik, 2021-09-23:
https://github.com/llvm/circt/blob/069faf81d8535556bb0dcd38b195017936976668/include/circt/Dialect/Calyx/CalyxStructure.td#L293-L295

IIRC, it would automatically parse the `guard` as the `src` when no guard was provided.

PR #1879 (2021-09-25): [Calyx] Use native compiler representation for guarded assignments and ports.
Switches the GroupGoOp, GroupDoneOp, and AssignOp to match the native compiler representation.

```mlir
// Before
calyx.assign %dest = %src, %guard ? : i8

calyx.group_done %src, %guard ? : i1 

%result = calyx.group_go %src, %guard ? : i1
```

```mlir
// After
calyx.assign %dest = %guard ? %src : i8

calyx.group_done %guard ? %src : i1

%result = calyx.group_go %guard ? %src : i1
```

I think Mike brought up a good point in #1873. We're still pretty dependent on the native compiler, so it seems right to keep the IRs [... cut]
commit b58fa876 (2021-09-25): [Calyx] Use native compiler representation for guarded assignments and ports. (#1879)

Switches the GroupGoOp, GroupDoneOp, and AssignOp to match the native compiler representation. 
We're still pretty dependent on the native compiler, so it seems right to keep the IRs as similar as 
possible. Closes #1873.
closed 2021-09-25 by cgyurgyik