llvm/circt #1878: expand whens with nested if not driving mux with conjunction of conditions

darthscsi, 2021-09-24:
expand whens with nested if not driving mux with conjunction of conditions.  It is, how ever computing it, just not using it.
From:
```
circuit CountAndDriver :
  module CountAndDriver :
    input clock : Clock
    input reset : UInt<1>
    output io : { flip default_value : UInt<1>, flip count : UInt<16>, driven_value : UInt<1>}

    clock is invalid
    reset is invalid
    io is invalid
    io.driven_value <= UInt<1>("h1") 
    reg counter : UInt<16>, clock with :
      reset => (reset, UInt<16>("h0")) 
    node _T = not(io.default_value) 
    when _T : 
      node _T_1 = lt(counter, io.count)
      when _T_1 : 
        node _counter_T = add(counter, UInt<1>("h1")) 
        node _counter_T_1 = tail(_counter_T, 1) 
        counter <= _counter_T_1 
    node _T_2 = leq(io.count, counter) 
    when _T_2 
      io.driven_value <= io.default_value
```

After some fiddling gives:
```
module  {
  firrtl.circuit "CountAndDriver"   {
    firrtl.module @CountAndDriver(in %clock: !firrtl.clock, in %reset: !firrtl.uint<1>, out %io: !firrtl.bundle<default_value flip: uint<1>, count flip: uint<16>, driven_value: uint<1>>) {
      %0 = firrtl.subfield %io(0) : (!firrtl.bundle<default_value flip: uint<1>, count flip: uint<16>, driven_value: uint<1>>) -> !firrtl.uint<1>
      %1 = firrtl.subfield %io(1) : (!firrtl.bundle<default_value flip: uint<1>, count flip: uint<16>, driven_value: uint<1>>) -> !firrtl.uint<16>
      %2 = firrtl.subfield %io(2) : (!firrtl.bundle<default_value flip: uint<1>, count flip: uint<16>, driven_value: uint<1>>) -> !firrtl.uint<1>
      %invalid_ui1 = firrtl.invalidvalue : !firrtl.uint<1>
      firrtl.connect %2, %invalid_ui1 : !firrtl.uint<1>, !firrtl.uint<1>
      %3 = firrtl.subfield %io(2) : (!firrtl.bundle<default_value flip: uint<1>, count flip: uint<16>, driven_value: uint<1>>) -> !firrtl.uint<1>
      %c1_ui1 = firrtl.constant 1 : !firrtl.uint<1>
      firrtl.connect %3, %c1_ui1 : !firrtl.uint<1>, !firrtl.uint<1>
      %c0_ui16 = firrtl.constant 0 : !firrtl.uint<16>
      %counter = firrtl.regreset %clock, %reset, %c0_ui16  : !firrtl.uint<1>, !firrtl.uint<16>, !firrtl.uint<16>
      %4 = firrtl.subfield %io(0) : (!firrtl.bundle<default_value flip: uint<1>, count flip: uint<16>, driven_value: uint<1>>) -> !firrtl.uint<1>
      %5 = firrtl.not %4 : (!firrtl.uint<1>) -> !firrtl.uint<1>
      %_T = firrtl.node %5  : !firrtl.uint<1>
      firrtl.when %_T  {
        %8 = firrtl.subfield %io(1) : (!firrtl.bundle<default_value flip: uint<1>, count flip: uint<16>, driven_value: uint<1>>) -> !firrtl.uint<16>
        %9 = firrtl.lt %counter, %8 : (!firrtl.uint<16>, !firrtl.uint<16>) -> !firrtl.uint<1>
        %_T_1 = firrtl.node %9  : !firrtl.uint<1>
        firrtl.when %_T_1  {
          %10 = firrtl.add %counter, %c1_ui1 : (!firrtl.uint<16>, !firrtl.uint<1>) -> !firrtl.uint<17>
          %_counter_T = firrtl.node %10  : !firrtl.uint<17>
          %11 = firrtl.tail %_counter_T, 1 : (!firrtl.uint<17>) -> !firrtl.uint<16>
          %_counter_T_1 = firrtl.node %11  : !firrtl.uint<16>
          firrtl.connect %counter, %_counter_T_1 : !firrtl.uint<16>, !firrtl.uint<16>
        }
      }
      %6 = firrtl.subfield %io(1) : (!firrtl.bundle<default_value flip: uint<1>, count flip: uint<16>, driven_value: uint<1>>) -> !firrtl.uint<16>
      %7 = firrtl.leq %6, %counter : (!firrtl.uint<16>, !firrtl.uint<16>) -> !firrtl.uint<1>
      %_T_2 = firrtl.node %7  : !firrtl.uint<1>
      firrtl.when %_T_2  {
        %8 = firrtl.subfield %io(2) : (!firrtl.bundle<default_value flip: uint<1>, count flip: uint<16>, driven_value: uint<1>>) -> !firrtl.uint<1>
        %9 = firrtl.subfield %io(0) : (!firrtl.bundle<default_value flip: uint<1>, count flip: uint<16>, driven_value: uint<1>>) -> !firrtl.uint<1>
        firrtl.connect %8, %9 : !firrtl.uint<1>, !firrtl.uint<1>
      }
    }
  }
}
```

After expand whens:

```
module  {
  firrtl.circuit "CountAndDriver"   {
    firrtl.module @CountAndDriver(in %clock: !firrtl.clock, in %reset: !firrtl.uint<1>, out %io: !firrtl.bundle<default_value flip: uint<1>, count flip: uint<16>, driven_value: uint<1>>) {
      %0 = firrtl.subfield %io(0) : (!firrtl.bundle<default_value flip: uint<1>, count flip: uint<16>, driven_value: uint<1>>) -> !firrtl.uint<1>
      %1 = firrtl.subfield %io(1) : (!firrtl.bundle<default_value flip: uint<1>, count flip: uint<16>, driven_value: uint<1>>) -> !firrtl.uint<16>
      %2 = firrtl.subfield %io(2) : (!firrtl.bundle<default_value flip: uint<1>, count flip: uint<16>, driven_value: uint<1>>) -> !firrtl.uint<1>
      %invalid_ui1 = firrtl.invalidvalue : !firrtl.uint<1>
      %3 = firrtl.subfield %io(2) : (!firrtl.bundle<default_value flip: uint<1>, count flip: uint<16>, driven_value: uint<1>>) -> !firrtl.uint<1>
      %c1_ui1 = firrtl.constant 1 : !firrtl.uint<1>
      %c0_ui16 = firrtl.constant 0 : !firrtl.uint<16>
      %counter = firrtl.regreset %clock, %reset, %c0_ui16  : !firrtl.uint<1>, !firrtl.uint<16>, !firrtl.uint<16>
      %4 = firrtl.subfield %io(0) : (!firrtl.bundle<default_value flip: uint<1>, count flip: uint<16>, driven_value: uint<1>>) -> !firrtl.uint<1>
      %5 = firrtl.not %4 : (!firrtl.uint<1>) -> !firrtl.uint<1>
      %_T = firrtl.node %5  : !firrtl.uint<1>
      %6 = firrtl.subfield %io(1) : (!firrtl.bundle<default_value flip: uint<1>, count flip: uint<16>, driven_value: uint<1>>) -> !firrtl.uint<16>
      %7 = firrtl.lt %counter, %6 : (!firrtl.uint<16>, !firrtl.uint<16>) -> !firrtl.uint<1>
      %_T_1 = firrtl.node %7  : !firrtl.uint<1>
      %8 = firrtl.and %_T, %_T_1 : (!firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<1>
      %9 = firrtl.add %counter, %c1_ui1 : (!firrtl.uint<16>, !firrtl.uint<1>) -> !firrtl.uint<17>
      %_counter_T = firrtl.node %9  : !firrtl.uint<17>
      %10 = firrtl.tail %_counter_T, 1 : (!firrtl.uint<17>) -> !firrtl.uint<16>
      %_counter_T_1 = firrtl.node %10  : !firrtl.uint<16>
      %11 = firrtl.mux(%_T, %_counter_T_1, %counter) : (!firrtl.uint<1>, !firrtl.uint<16>, !firrtl.uint<16>) -> !firrtl.uint<16>
      firrtl.connect %counter, %11 : !firrtl.uint<16>, !firrtl.uint<16>
      %12 = firrtl.subfield %io(1) : (!firrtl.bundle<default_value flip: uint<1>, count flip: uint<16>, driven_value: uint<1>>) -> !firrtl.uint<16>
      %13 = firrtl.leq %12, %counter : (!firrtl.uint<16>, !firrtl.uint<16>) -> !firrtl.uint<1>
      %_T_2 = firrtl.node %13  : !firrtl.uint<1>
      %14 = firrtl.subfield %io(2) : (!firrtl.bundle<default_value flip: uint<1>, count flip: uint<16>, driven_value: uint<1>>) -> !firrtl.uint<1>
      %15 = firrtl.subfield %io(0) : (!firrtl.bundle<default_value flip: uint<1>, count flip: uint<16>, driven_value: uint<1>>) -> !firrtl.uint<1>
      %16 = firrtl.mux(%_T_2, %15, %c1_ui1) : (!firrtl.uint<1>, !firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<1>
      firrtl.connect %14, %16 : !firrtl.uint<1>, !firrtl.uint<1>
    }
  }
}

```

Note that %11 should be conditioned on %8, not %_T

youngar, 2021-09-25:
Yeah, this is really busted. With the previous change we're not carrying-forward the condition of the outer mux (https://github.com/llvm/circt/pull/1585/files).  Right now the code is tracking only 1 level of scoping of previous values, when it needs to carry all of them.  I'll have think about how I want to solve this, I might rearchitect some of the code instead of dropping in something like a `ScopedHashtable`.  A more minimal testcast is:

```
circuit CountAndDriver :
  module CountAndDriver :
    input reset0 : UInt<1>
    input reset1 : UInt<1>
    input in0 : UInt<4>
    input in1 : UInt<4>
    output out1 : UInt<4>

    out1 <= in0
    when reset0 :
      when reset1 :
        out1 <= in1
 ```
 ```mlir
firrtl.module @CountAndDriver(in %clock: !firrtl.clock, in %reset0: !firrtl.uint<1>, in %reset1: !firrtl.uint<1>, in %in0: !firrtl.uint<4>, in %in1: !firrtl.uint<4>, in %in2: !firrtl.uint<4>, out %out1: !firrtl.uint<4>) {
  firrtl.connect %out1, %in0 : !firrtl.uint<4>, !firrtl.uint<4>
  firrtl.when %reset0  {
    firrtl.when %reset1  {
      firrtl.connect %out1, %in1 : !firrtl.uint<4>, !firrtl.uint<4>
    }
  }
}

// -----// IR Dump After ExpandWhens //----- //
firrtl.module @CountAndDriver(in %clock: !firrtl.clock, in %reset0: !firrtl.uint<1>, in %reset1: !firrtl.uint<1>, in %in0: !firrtl.uint<4>, in %in1: !firrtl.uint<4>, in %in2: !firrtl.uint<4>, out %out1: !firrtl.uint<4>) {
  %0 = firrtl.and %reset0, %reset1 : (!firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<1>
  %1 = firrtl.mux(%reset0, %in1, %in0) : (!firrtl.uint<1>, !firrtl.uint<4>, !firrtl.uint<4>) -> !firrtl.uint<4>
  firrtl.connect %out1, %1 : !firrtl.uint<4>, !firrtl.uint<4>
}
```

closed 2021-09-27 by youngar

youngar, 2021-09-27:
With the fix we are now producing the following from MFC:
```verilog
module CountAndDriver(	// ./test.fir:2:10
  input        reset0, reset1,
  input  [3:0] in0, in1,
  output [3:0] out1);

  assign out1 = reset0 ? (reset1 ? in1 : in0) : in0;	// ./test.fir:2:10, :12:14
endmodule
```
And SFC:
```verilog
// andrewy@ayoung-0504~/w/c/build-release> ~/wsp/firrtl-1.5-SNAPSHOT -i test.fir -o test

module CountAndDriver(
  input        reset0,
  input        reset1,
  input  [3:0] in0,
  input  [3:0] in1,
  output [3:0] out1
);
  wire [3:0] _GEN_0 = reset1 ? in1 : in0; // @[]
  assign out1 = reset0 ? _GEN_0 : in0; // @[]
endmodule
```

commit 728d63f7 (2021-09-27): [FIRRTL][ExpandWhens] Properly track all outer scope value (#1882)

When ExpandWhens sees the the following code, it *should* produce the
following:

```firrtl
out <= in0
when p0 :
  when p1 :
    out <= in1
```

```firrtl
out <= mux(p0, mux(p1, in1, in0), in0)
```

There was a mistake in the code where it would only track previously
driven values from one level of outer scope.  This means that when
producing the inner mux, `mux(p1, in1, in0)`, it never found that `in0`
was previously driven to `out`, and never formed a mux with the `in1`
connection:

```firrtl
out <= mux(p0, in1, in0)
```

To [... cut]