llvm/circt #1925: Undefined reference for mlir::Pass

cgyurgyik, 2021-10-04:
```
/usr/bin/ld: CMakeFiles/obj.CIRCTExportVerilog.dir/ExportVerilog.cpp.o: in function `mlir::Pass::~Pass()':
ExportVerilog.cpp:(.text._ZN4mlir4PassD2Ev[_ZN4mlir4PassD2Ev]+0xb): undefined reference to `vtable for mlir::Pass'
/usr/bin/ld: CMakeFiles/obj.CIRCTExportVerilog.dir/ExportVerilog.cpp.o:(.data.rel.ro._ZTVN12_GLOBAL__N_121ExportVerilogFilePassE+0x40): undefined reference to `mlir::Pass::initializeOptions(llvm::StringRef)'
/usr/bin/ld: CMakeFiles/obj.CIRCTExportVerilog.dir/ExportVerilog.cpp.o:(.data.rel.ro._ZTVN12_GLOBAL__N_121ExportVerilogFilePassE+0x60): undefined reference to `mlir::Pass::anchor()'
/usr/bin/ld: CMakeFiles/obj.CIRCTExportVerilog.dir/ExportVerilog.cpp.o:(.data.rel.ro._ZTVN12_GLOBAL__N_122ExportSplitVerilogPassE+0x40): undefined reference to `mlir::Pass::initializeOptions(llvm::StringRef)'
/usr/bin/ld: CMakeFiles/obj.CIRCTExportVerilog.dir/ExportVerilog.cpp.o:(.data.rel.ro._ZTVN12_GLOBAL__N_122ExportSplitVerilogPassE+0x60): undefined reference to `mlir::Pass::anchor()'
/usr/bin/ld: CMakeFiles/obj.CIRCTExportVerilog.dir/ExportVerilog.cpp.o:(.data.rel.ro._ZTVN5circt12translations26ExportSplitVerilogPassBaseIN12_GLOBAL__N_122ExportSplitVerilogPassEEE+0x40): undefined reference to `mlir::Pass::initializeOptions(llvm::StringRef)'
/usr/bin/ld: CMakeFiles/obj.CIRCTExportVerilog.dir/ExportVerilog.cpp.o:(.data.rel.ro._ZTVN5circt12translations26ExportSplitVerilogPassBaseIN12_GLOBAL__N_122ExportSplitVerilogPassEEE+0x60): undefined reference to `mlir::Pass::anchor()'
/usr/bin/ld: CMakeFiles/obj.CIRCTExportVerilog.dir/ExportVerilog.cpp.o:(.data.rel.ro._ZTVN4mlir4Pass6OptionINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEN4llvm2cl6parserIS7_EEEE[_ZTVN4mlir4Pass6OptionINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEN4llvm2cl6parserIS7_EEEE]+0xa0): undefined reference to `mlir::detail::PassOptions::OptionBase::anchor()'
/usr/bin/ld: CMakeFiles/obj.CIRCTExportVerilog.dir/ExportVerilog.cpp.o:(.data.rel.ro._ZTVN4mlir6detail11PassOptions6OptionINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEN4llvm2cl6parserIS8_EEEE[_ZTVN4mlir6detail11PassOptions6OptionINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEN4llvm2cl6parserIS8_EEEE]+0xa0): undefined reference to `mlir::detail::PassOptions::OptionBase::anchor()'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[3]: *** [lib/Translation/ExportVerilog/CMakeFiles/CIRCTExportVerilog.dir/build.make:85: lib/libCIRCTExportVerilog.so.14git] Error 1
make[2]: *** [CMakeFiles/Makefile2:9454: lib/Translation/ExportVerilog/CMakeFiles/CIRCTExportVerilog.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
[ 88%] Built target obj.CIRCTCAPISV
make[1]: *** [CMakeFiles/Makefile2:10637: test/CMakeFiles/check-circt.dir/rule] Error 2
make: *** [Makefile:4038: check-circt] Error 2
```

cgyurgyik, 2021-10-04:
Latest run: https://github.com/llvm/circt/runs/3796733517

richardxia, 2021-10-04:
I started looking at this, and thanks to a few tips from @youngar, it looks like you need to set the `BUILD_SHARED_LIBS=ON` option in order to reproduce locally.

closed 2021-10-04 by richardxia

cgyurgyik, 2021-10-04:
Cool! thanks for fixing this.

richardxia, 2021-10-04:
No problem!

commit 3242f0bf (2021-10-04): [ExportVerilog] Add MLIRPass to CMakeLists.txt.

This should fix the build errors in main, where we fail to link to some
symbols defined in MLIRPass. This build error appears to only manifest
when BUILD_SHARED_LIBS=ON.

Fixes #1925.
PR #1926 (2021-10-04): [ExportVerilog] Add MLIRPass to CMakeLists.txt.
~Marking this as a Draft PR since I haven't fully tested it locally. I'll update this when I confirm that it passes for me locally.~

This should fix the build errors in main, where we fail to link to some
symbols defined in MLIRPass. This build error appears to only manifest
when BUILD_SHARED_LIBS=ON.

Fixes #1925.
commit 8dca4c38 (2021-10-04): [ExportVerilog] Add MLIRPass to CMakeLists.txt. (#1926)

This should fix the build errors in main, where we fail to link to some
symbols defined in MLIRPass. This build error appears to only manifest
when BUILD_SHARED_LIBS=ON.

Fixes #1925.