llvm/circt #1932: [CI] [ExportVerilog]

teqdruid, 2021-10-05:
Seems like the ExportVerilog pass is producing incorrect Verilog: https://github.com/llvm/circt/runs/3805588305?check_suite_focus=true

```
********************
Testing:  0.. 10.. 20.. 30.. 40.. 50.. 60.
FAIL: CIRCT :: EmitVerilog/basic.mlir (20 of 28)
******************** TEST 'CIRCT :: EmitVerilog/basic.mlir' FAILED ********************
Script:
--
: 'RUN: at line 2';   /__w/circt/circt/build/bin/circt-opt /__w/circt/circt/integration_test/EmitVerilog/basic.mlir -export-verilog -verify-diagnostics > /__w/circt/circt/build/tools/circt/integration_test/EmitVerilog/Output/basic.mlir.tmp1.sv
: 'RUN: at line 3';   /__w/circt/circt/build/bin/circt-rtl-sim.py /__w/circt/circt/build/tools/circt/integration_test/EmitVerilog/Output/basic.mlir.tmp1.sv --cycles 8 2>&1 | /__w/circt/circt/build/bin/FileCheck /__w/circt/circt/integration_test/EmitVerilog/basic.mlir
--
Exit Code: 1

Command Output (stderr):
--
/__w/circt/circt/integration_test/EmitVerilog/basic.mlir:27:11: error: CHECK: expected string not found in input
// CHECK: [driver] Starting simulation
          ^
<stdin>:1:1: note: scanning from here
%Error: /__w/circt/circt/build/tools/circt/integration_test/EmitVerilog/Output/basic.mlir.tmp1.sv:31:9: syntax error, unexpected '{', expecting IDENTIFIER or TYPE-IDENTIFIER or randomize
^
<stdin>:4:149: note: possible intended match here
%Error-UNSUPPORTED: /__w/circt/circt/build/tools/circt/integration_test/EmitVerilog/Output/basic.mlir.tmp1.sv:35:17: Unsupported: Verilog 2001-config reserved word not implemented: 'instance'
                                                                                                                                                    ^

Input file: <stdin>
Check file: /__w/circt/circt/integration_test/EmitVerilog/basic.mlir

-dump-input=help explains the following input dump.

Input was:
<<<<<<
            1: %Error: /__w/circt/circt/build/tools/circt/integration_test/EmitVerilog/Output/basic.mlir.tmp1.sv:31:9: syntax error, unexpected '{', expecting IDENTIFIER or TYPE-IDENTIFIER or randomize 
check:27'0     X~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ error: no match found
            2:  31 | module { 
check:27'0     ~~~~~~~~~~~~~~~
            3:  | ^ 
check:27'0     ~~~~~
            4: %Error-UNSUPPORTED: /__w/circt/circt/build/tools/circt/integration_test/EmitVerilog/Output/basic.mlir.tmp1.sv:35:17: Unsupported: Verilog 2001-config reserved word not implemented: 'instance' 
check:27'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
check:27'1                                                                                                                                                         ?                                            possible intended match
            5:  35 | %aaa.f = hw.instance "aaa" @AAA() -> (f: i1) 
check:27'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            6:  | ^~~~~~~~ 
check:27'0     ~~~~~~~~~~~~
            7: %Error-UNSUPPORTED: /__w/circt/circt/build/tools/circt/integration_test/EmitVerilog/Output/basic.mlir.tmp1.sv:38:17: Unsupported: Verilog 2001-config reserved word not implemented: 'instance' 
check:27'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            8:  38 | %shl.b = hw.instance "shl" @shl(a: %0: i1) -> (b: i1) 
check:27'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            9:  | ^~~~~~~~ 
check:27'0     ~~~~~~~~~~~~
            .
            .
            .
>>>>>>

--

********************
Testing:  0.. 10.. 20.. 30.. 40.. 50.. 60.. 70
FAIL: CIRCT :: EmitVerilog/lint.mlir (22 of 28)
******************** TEST 'CIRCT :: EmitVerilog/lint.mlir' FAILED ********************
Script:
--
: 'RUN: at line 2';   /__w/circt/circt/build/bin/circt-opt /__w/circt/circt/integration_test/EmitVerilog/lint.mlir -export-verilog -verify-diagnostics > /__w/circt/circt/build/tools/circt/integration_test/EmitVerilog/Output/lint.mlir.tmp1.sv
: 'RUN: at line 3';   /usr/bin/verilator --lint-only --top-module A /__w/circt/circt/build/tools/circt/integration_test/EmitVerilog/Output/lint.mlir.tmp1.sv
: 'RUN: at line 4';   /usr/bin/verilator --lint-only --top-module AB /__w/circt/circt/build/tools/circt/integration_test/EmitVerilog/Output/lint.mlir.tmp1.sv
: 'RUN: at line 5';   /usr/bin/verilator --lint-only --top-module shl /__w/circt/circt/build/tools/circt/integration_test/EmitVerilog/Output/lint.mlir.tmp1.sv
: 'RUN: at line 6';   /usr/bin/verilator --lint-only --top-module TESTSIMPLE /__w/circt/circt/build/tools/circt/integration_test/EmitVerilog/Output/lint.mlir.tmp1.sv
: 'RUN: at line 7';   /usr/bin/verilator --lint-only --top-module casts /__w/circt/circt/build/tools/circt/integration_test/EmitVerilog/Output/lint.mlir.tmp1.sv
: 'RUN: at line 8';   /usr/bin/verilator --lint-only --top-module exprInlineTestIssue439 /__w/circt/circt/build/tools/circt/integration_test/EmitVerilog/Output/lint.mlir.tmp1.sv
: 'RUN: at line 9';   /usr/bin/verilator --lint-only --top-module StructDecls /__w/circt/circt/build/tools/circt/integration_test/EmitVerilog/Output/lint.mlir.tmp1.sv
--
Exit Code: 1

Command Output (stderr):
--
%Error: /__w/circt/circt/build/tools/circt/integration_test/EmitVerilog/Output/lint.mlir.tmp1.sv:121:9: syntax error, unexpected '{', expecting IDENTIFIER or TYPE-IDENTIFIER or randomize
  121 | module  {
      |         ^
%Error-UNSUPPORTED: /__w/circt/circt/build/tools/circt/integration_test/EmitVerilog/Output/lint.mlir.tmp1.sv:139:16: Unsupported: Verilog 2001-config reserved word not implemented: 'instance'
  139 |     %a1.f = hw.instance "a1" @AAA(d: %w: i1, e: %0: i1) -> (f: i1)
      |                ^~~~~~~~
%Error-UNSUPPORTED: /__w/circt/circt/build/tools/circt/integration_test/EmitVerilog/Output/lint.mlir.tmp1.sv:142:23: Unsupported: Verilog 2001-config reserved word not implemented: 'instance'
  142 |     %b1.b, %b1.c = hw.instance "b1" @B(a: %1: i1) -> (b: i1, c: i1)
      |                       ^~~~~~~~
%Error: Exiting due to 3 error(s)

--

********************
Testing:  0.. 10.. 20.. 30.. 40.. 50.. 60.. 70.. 80.. 90.. 
********************
Failed Tests (5):
  CIRCT :: ESI/cosim/basic.mlir
  CIRCT :: ESI/cosim/loopback.mlir
  CIRCT :: ESI/system/basic.mlir
  CIRCT :: EmitVerilog/basic.mlir
  CIRCT :: EmitVerilog/lint.mlir

Testing Time: 7.30s
  Unsupported:  7
  Passed     : 16
  Failed     :  5
```

lattner, 2021-10-05:
export-verilog is now a pass and `circt-opt` uses it in the default mode which prints to standard out.  However, it *also* prints the output IR to standard out as well by default.  Can you try changing this:

`circt-opt %s -export-verilog -verify-diagnostics > %t1.sv`

to:

`circt-opt %s -export-verilog -verify-diagnostics -o %t2.mlir > %t1.sv`

?

teqdruid, 2021-10-05:
Thanks @lattner : e7990b8a65421f184dd3600600e34a9b137cb1ea

Passed locally, re-running integration tests now: https://github.com/llvm/circt/actions/runs/1309538973

lattner, 2021-10-05:
Awesome, thank you John!

teqdruid, 2021-10-06:
Looks like the GH run will pass.

closed 2021-10-06 by teqdruid