llvm/circt #2006: Adopt "pretty accessors" in MLIR

lattner, 2021-10-16:
MLIR mainline added the ability to synthesize get/setters with a properly camel cased and get/set-prefixed name, see [this patch](https://reviews.llvm.org/D111033) and [this announcement](https://llvm.discourse.group/t/psa-ods-generated-accessors-will-change-to-have-a-get-prefix-update-you-apis/4476/2).

We should do a merge from MLIR mainline to get this, and then adopt it in the CIRCT dialects.  I don't think we need to do the multistage adoption thing, we can just jump directly to `_Prefixed`

lattner, 2021-10-24:
I think the Merge was done, but we haven't adopted the new names yet.

maerhart, 2022-07-14:
Dialects that have adopted prefixed accessors so far:
- [ ] Calyx
- [ ] Comb
- [ ] ESI
- [ ] FIRRTL
- [ ] FSM
- [ ] Handshake
- [ ] HW
- [ ] HWArith
- [x] LLHD
- [x] Moore
- [ ] MSFT
- [ ] Seq
- [ ] StaticLogic
- [ ] SV