llvm/circt #2073: Tabs in Verilog emission

drom, 2021-11-03:
Verilog emitter uses tabs in some places to align comments. Example

In port list:

[image]

In assign statements:

[image]

Tab size/behavior heavily depend on the editor / viewer and in general don't guaranty alignment.
We use spaces for indentation and alignment in most of the cases.
Using tabs in just one particular case and in the in the middle of the line complicates handling of the code.

## How to reproduce

The following FIRRTL program

```python
circuit top_mod:
  module top_mod:
    input a: UInt<42>
    output b: UInt
    output bb: UInt
    output bbb: UInt
    output bbbb: UInt
    b <= tail(a, 40)
    bb <= tail(a, 3)
    bbb <= tail(a, 2)
    bbbb <= tail(a, 1)
```

Compiled with `firtool   --lower-to-hw --infer-widths --imconstprop --lowering-options=disallowPackedArrays,emittedLineLength=8192 --mlir-timing --verilog -o=$VFILE2` produces this code:

```verilog
module top_mod(	// a_top_mod.fir:2:10
  input  [41:0] a,
  output [1:0]  b,
  output [38:0] bb,
  output [39:0] bbb,
  output [40:0] bbbb);

  assign b = a[1:0];	// a_top_mod.fir:2:10, :8:10
  assign bb = a[38:0];	// a_top_mod.fir:2:10, :9:11
  assign bbb = a[39:0];	// a_top_mod.fir:2:10, :10:12
  assign bbbb = a[40:0];	// a_top_mod.fir:2:10, :11:13
endmodule
```

lattner, 2021-11-03:
Are you sure?  I'm quite certain we don't emit tabs before assign statements (maybe in some verbatim nodes it is possible?).  The example above you include isn't using tabs.

lattner, 2021-11-03:
Oh I see what you mean, we do use tabs before the `// location.fir` comment.

seldridge, 2021-11-03:
It's using tabs _after_ the assign and before the comment. Something like:

```
<space><space>assign foo = bar;<tab>// hello
```