llvm/circt #2084: Extract Test Code doesn't handle reads of output ports

darthscsi, 2021-11-04:
This produces incorrect verilog.  Not in the logic, per-se, but in the names being used by instances.

```  firrtl.circuit "SiFive_MulDiv" {
    firrtl.module @SiFive_MulDiv(in %clock: !firrtl.clock, in %reset: !firrtl.asyncreset, out %io_req_ready: !firrtl.uint<1>) {
      %c0_ui1 = firrtl.constant 0 : !firrtl.uint<1>
      %21 = firrtl.asUInt %reset : (!firrtl.asyncreset) -> !firrtl.uint<1>
      firrtl.assert %clock, %io_req_ready, %21, "blah"  {eventControl = 0 : i32, isConcurrent = true, name = "chisel3_builtin"}
      firrtl.connect %io_req_ready, %c0_ui1 : !firrtl.uint<1>, !firrtl.uint<1>
    }
}
```

darthscsi, 2021-11-04:
To check: `firtool --verilog  --extract-test-code |grep "\.\."`

prithayan, 2021-11-09:
The issue as pointed out by @fabianschuiki is that the `emitBind` function is not calling `getPortVerilogName`  to get the updated port names.
Here is this same test case 

```
module  {
  hw.module @SiFive_MulDiv_assert(%.io_req_ready.output: i1, %reset: i1, %clock: i1) {
    %true = hw.constant true
    %0 = comb.xor %reset, %true : i1
    %1 = comb.or %0, %.io_req_ready.output : i1
    sv.assert.concurrent posedge %clock, %1 label "assert__chisel3_builtin" message "blah"
    hw.output
  }
  hw.module @SiFive_MulDiv_assume(%.io_req_ready.output: i1, %reset: i1, %clock: i1) {
    sv.ifdef "USE_PROPERTY_AS_CONSTRAINT"  {
      %true = hw.constant true
      %0 = comb.xor %reset, %true : i1
      %1 = comb.or %0, %.io_req_ready.output : i1
      sv.assume.concurrent posedge %clock, %1 label "assume__chisel3_builtin"
    }
    hw.output
  }
  hw.module @SiFive_MulDiv(%clock: i1, %reset: i1) -> (io_req_ready: i1) {
    %false = hw.constant false
    hw.instance "InvisibleBind_assert" sym @__ETC_SiFive_MulDiv_assert @SiFive_MulDiv_assert(".io_req_ready.output": %false: i1, reset: %reset: i1, clock: %clock: i1) -> () {doNotPrint = true}
    hw.instance "InvisibleBind_assume" sym @__ETC_SiFive_MulDiv_assume @SiFive_MulDiv_assume(".io_req_ready.output": %false: i1, reset: %reset: i1, clock: %clock: i1) -> () {doNotPrint = true}
    hw.output %false : i1
  }
  sv.bind #hw.innerNameRef<@SiFive_MulDiv::@__ETC_SiFive_MulDiv_assert>
  sv.bind #hw.innerNameRef<@SiFive_MulDiv::@__ETC_SiFive_MulDiv_assume>
} 
```
After `export-verilog` this is the incorrect verilog. The updated module `SiFive_MulDiv_assert` port name  `_io_req_ready_output` is not reflected on the `bind` (`.io_req_ready.output`)  
```
module SiFive_MulDiv_assert(  // test1.mlir:2:3
  input _io_req_ready_output, reset, clock);

  assert__chisel3_builtin: assert property (@(posedge clock) ~reset | _io_req_ready_output) else $error("blah");  // test1.mlir:4:10, :5:10, :6:5
endmodule

module SiFive_MulDiv_assume(  // test1.mlir:9:3
  input _io_req_ready_output, reset, clock);

  `ifdef USE_PROPERTY_AS_CONSTRAINT // test1.mlir:10:5
    assume__chisel3_builtin: assume property (@(posedge clock) ~reset | _io_req_ready_output);  // test1.mlir:12:12, :13:12, :14:7
  `endif
endmodule

module SiFive_MulDiv( // test1.mlir:18:3
  input  clock, reset,
  output io_req_ready);

  wire InvisibleBind_assume__io_req_ready_output; // test1.mlir:21:5
  wire InvisibleBind_assert__io_req_ready_output; // test1.mlir:20:5

  assign InvisibleBind_assert__io_req_ready_output = 1'h0;  // test1.mlir:19:14, :20:5
  /* This instance is elsewhere emitted as a bind statement.
    SiFive_MulDiv_assert InvisibleBind_assert ( // test1.mlir:20:5
      ._io_req_ready_output (InvisibleBind_assert__io_req_ready_output),  // test1.mlir:20:5
      .reset                (reset),
      .clock                (clock)
    );
  */
  assign InvisibleBind_assume__io_req_ready_output = 1'h0;  // test1.mlir:19:14, :21:5
  /* This instance is elsewhere emitted as a bind statement.
    SiFive_MulDiv_assume InvisibleBind_assume ( // test1.mlir:21:5
      ._io_req_ready_output (InvisibleBind_assume__io_req_ready_output),  // test1.mlir:21:5
      .reset                (reset),
      .clock                (clock)
    );
  */
  assign io_req_ready = 1'h0; // test1.mlir:19:14, :22:5
endmodule

// ----- 8< ----- FILE "bindfile" ----- 8< -----

bind SiFive_MulDiv SiFive_MulDiv_assert InvisibleBind_assert (
  ..io_req_ready.output (InvisibleBind_assert_.io_req_ready.output),
  .reset                (reset),
  .clock                (clock)
);
bind SiFive_MulDiv SiFive_MulDiv_assume InvisibleBind_assume (
  ..io_req_ready.output (InvisibleBind_assume_.io_req_ready.output),
  .reset                (reset),
  .clock                (clock)
);
```

commit efd420ec (2021-11-09): [ExportVerilog] Get legalized bind port names (#2116)

The emitBind was using incorrect port names during ExportVerilog.
Use getPortVerilogName to get the updated name.
Fixes #2084
closed 2021-11-09 by prithayan