llvm/circt #2087: [ExportVerilog] Width warning about arrayg_get + mux

uenoku, 2021-11-05:
```verilog
hw.module @Foo(%array: !hw.array<4xi6>, %t: i1, %c:i1) -> (b: i6) {
    %true = hw.constant true
    %not = comb.xor %true, %t : i1 
    %concat = comb.concat %true, %not : i1, i1
    %h2 = hw.constant 0 : i2
    %index = comb.mux %c, %h2, %concat : i2
    %res = hw.array_get %array[%index] : !hw.array<4xi6>
    hw.output %res: i6
} 
```
will be lowered into 
```
module Foo(   
  input  [3:0][5:0] array,
  input             t, c,
  output [5:0]      b);

  assign b = array[c ? 2'h0 : {1'h1, 1'h1 ^ t}];  
endmodule
```

Verilator produces the following warnings
```
%Warning-WIDTH: a.sv:6:22: Operator COND expects 32 or 3 bits on the Conditional True, but Conditional True's CONST '2'h0' generates 2 bits.
                         : ... In instance Foo
    6 |   assign b = array[c ? 2'h0 : {1'h1, 1'h1 ^ t}];  
      |                      ^
                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.
%Warning-WIDTH: a.sv:6:22: Operator COND expects 32 or 3 bits on the Conditional False, but Conditional False's REPLICATE generates 2 bits.
                         : ... In instance Foo
    6 |   assign b = array[c ? 2'h0 : {1'h1, 1'h1 ^ t}];  
      |       
```

uenoku, 2021-11-05:
These two don't produce warnings. Hmm.. 
```verilog
  wire [1:0] tmp = c ? 2'h0 : d;
  assign b = array[tmp];
```
```verilog
  assign b = array[$unsigned(c ? 2'h0 : d)];  
```

seldridge, 2021-11-05:
This looks like a bug in Verilator: https://github.com/verilator/verilator/issues/1613

For now we can just ignore this (eventually we can rely on CIRCT to generate legal Verilog even if it violates incorrect lint rules in a downstream tool).  Eventually, this is an interesting case to explore:

1. Per-tool options for Verilog emission. E.g., if we had a `--tool verilator` option, spilling expressions that access packed arrays could always be spilled to temporaries would be one of the lowering options that gets turned on.
2. Similar to above, but automatic generation of lint waivers (either inline or in an external file) that work with the tools the user tells us they're going to use.

If a user _really_ cares about this, they can use the `disallowPackedarrays` option to disable the common situation where CIRCT will generate this style of Verilog. However, a user can still hit this from legal HW dialect. 😬

uenoku, 2021-11-05:
Fair point. Let's leave it as it is 😃