llvm/circt #2091: [LowerToHW] Verification of async reset value is not correct.

uenoku, 2021-11-07:
`test/Conversion/FIRRTLToHW/lower-to-hw.mlir` is currently missing `-verify-diagnostics` flag and producing the following unexpected errors.
```
/home/uenoku/dev/circt/test/Conversion/FIRRTLToHW/lower-to-hw.mlir:1144:11: error: register with async reset requires constant reset value
    %r0 = firrtl.regreset %clock, %arst, %constWire : !firrtl.asyncreset, !firrtl.uint<42>, !firrtl.uint<42>
          ^
/home/uenoku/dev/circt/test/Conversion/FIRRTLToHW/lower-to-hw.mlir:1144:11: note: see current operation: %10 = "firrtl.regreset"(%2, %3, %8) {annotations = [], name = "r0"} : (!firrtl.clock, !firrtl.asyncreset, !firrtl.uint<42>) -> !firrtl.uint<42>
/home/uenoku/dev/circt/test/Conversion/FIRRTLToHW/lower-to-hw.mlir:1140:18: note: reset value defined here:
    %constWire = firrtl.wire : !firrtl.uint<42>
                 ^
/home/uenoku/dev/circt/test/Conversion/FIRRTLToHW/lower-to-hw.mlir:1144:11: error: 'firrtl.regreset' op LowerToHW couldn't handle this operation
    %r0 = firrtl.regreset %clock, %arst, %constWire : !firrtl.asyncreset, !firrtl.uint<42>, !firrtl.uint<42>
          ^
/home/uenoku/dev/circt/test/Conversion/FIRRTLToHW/lower-to-hw.mlir:1144:11: note: see current operation: %10 = "firrtl.regreset"(%2, %3, %8) {annotations = [], name = "r0"} : (!firrtl.clock, !firrtl.asyncreset, !firrtl.uint<42>) -> !firrtl.uint<42>
/home/uenoku/dev/circt/test/Conversion/FIRRTLToHW/lower-to-hw.mlir:1152:11: error: register with async reset requires constant reset value
    %r0 = firrtl.regreset %clock, %arst, %constNode : !firrtl.asyncreset, !firrtl.uint<42>, !firrtl.uint<42>
          ^
/home/uenoku/dev/circt/test/Conversion/FIRRTLToHW/lower-to-hw.mlir:1152:11: note: see current operation: %6 = "firrtl.regreset"(%1, %2, %4) {annotations = [], name = "r0"} : (!firrtl.clock, !firrtl.asyncreset, !firrtl.uint<42>) -> !firrtl.uint<42>
/home/uenoku/dev/circt/test/Conversion/FIRRTLToHW/lower-to-hw.mlir:1150:18: note: reset value defined here:
    %constNode = firrtl.node %c1337_ui42 : !firrtl.uint<42>
                 ^
/home/uenoku/dev/circt/test/Conversion/FIRRTLToHW/lower-to-hw.mlir:1152:11: error: 'firrtl.regreset' op LowerToHW couldn't handle this operation
    %r0 = firrtl.regreset %clock, %arst, %constNode : !firrtl.asyncreset, !firrtl.uint<42>, !firrtl.uint<42>
          ^
/home/uenoku/dev/circt/test/Conversion/FIRRTLToHW/lower-to-hw.mlir:1152:11: note: see current operation: %6 = "firrtl.regreset"(%1, %2, %4) {annotations = [], name = "r0"} : (!firrtl.clock, !firrtl.asyncreset, !firrtl.uint<42>) -> !firrtl.uint<42>
``` 
This verification is introduced in https://github.com/llvm/circt/pull/1444 and it seems `firrtl::isConstant` doesn't work well for constant wire/node.  It is necessary to care about the direction of data dependency in the worklist iterations of `firrtl::isConstant`.

fabianschuiki, 2021-11-08:
Excellent catch @uenoku! I think the chain here is that the code that checks for const reset values of async reset registers inside `LowerToHW` implicitly assumes that constant folders and propagation of const values through wires has happened. It's doing a pretty dumb check for constness. Technically, it would have to look through wires and connects, resolve last connect semantics and partial connects / connects to subfields, and then see if the value is constant.

I'm not quite sure what the best way forward is. The errors are false positives in a sense, and it may be a bit weird that we rely on certain canonicalizations to happen to get correct output. On the other hand, if this gets lowered to HW and the wire is still there, the downstream synthesizer might not pick up on the reset value being a constant -- so you could also argue that this error triggering is actually correct. Maybe the underlying issue is that these checks are in the `lower-to-hw.mlir` test, where they should actually go into an integration-style test which runs both const prop and LowerToHW. Any thoughts, @darthscsi, @seldridge?

uenoku, 2021-11-08:
> you could also argue that this error triggering is actually correct

Yes, it makes sense. It would be accurate to say "the verification might be pessimistic"😃

seldridge, 2021-11-08:
> Maybe the underlying issue is that these checks are in the lower-to-hw.mlir test, where they should actually go into an integration-style test which runs both const prop and LowerToHW.

I agree with this. I think `lower-to-hw.mlir` needs to be updated to _not_ do constant initialization through a wire/node. We should then have integration tests that are checking this behavior.