llvm/circt #2093: [InferWidths] firtool crashes on ICache.fir in firrtl regress suite

lattner, 2021-11-07:
$ firtool -verilog ICache.fir 
ICache.fir:85:14: error: 'firrtl.mux' op LowerToHW couldn't handle this operation
      _T_287 <= _T_282
             ^
ICache.fir:85:14: note: see current operation: %360 = "firrtl.mux"(%357, %353, %66) : (!firrtl.uint<1>, !firrtl.uint<6>, !firrtl.uint) -> !firrtl.uint
ICache.fir:85:14: error: 'firrtl.connect' op LowerToHW couldn't handle this operation
      _T_287 <= _T_282
             ^
ICache.fir:85:14: note: see current operation: "firrtl.connect"(%359, %360) : (!firrtl.uint<6>, !firrtl.uint) -> ()
ICache.fir:324:14: error: 'firrtl.mux' op LowerToHW couldn't handle this operation
      _T_562 <= _T_557
             ^
...
ICache.fir:402:14: note: see current operation: "firrtl.connect"(%748, %749) : (!firrtl.uint<9>, !firrtl.uint) -> ()
Assertion failed: (entry != nameTable.end() && "value expected a name but doesn't have one"), function getName, file ExportVerilog.cpp, line 480.
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump without symbol names (ensure you have llvm-symbolizer in your PATH or set the environment var `LLVM_SYMBOLIZER_PATH` to point to it):
0  firtool                  0x000000010a8c06a7 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) + 39
1  firtool                  0x000000010a8bf4c8 llvm::sys::RunSignalHandlers() + 248
2  firtool                  0x000000010a8c0cf0 SignalHandler(int) + 288
3  libsystem_platform.dylib 0x00007fff2049ad7d _sigtramp + 29
4  libsystem_platform.dylib 0x000070000733a348 _sigtramp + 18446726485397599720
5  libsystem_c.dylib        0x00007fff203aa406 abort + 125
6  libsystem_c.dylib        0x00007fff203a97d8 err + 0
7  firtool                  0x000000010acb2bd3 (anonymous namespace)::ModuleNameManager::getName(llvm::PointerUnion<mlir::Value, mlir::Operation*>) (.cold.1) +

lattner, 2021-11-07:
InferWidths should emit an error if something can't be inferred.  And ExportVerilog should never crash.

youngar, 2021-11-07:
This looks like the problem we're seeing on a bunch of fir files since the last LLVM bump. I tracked it back to this commit in LLVM: https://reviews.llvm.org/D113138. I haven't sorted out what is going wrong here, but `InferWidths` normally emits errors when it can't infer a width. 

After all those errors print from`LowerToHW`, the pass does not fail, and firtool keeps running more passes.  Eventually it is asserting in `ExportVerilog` with `"value expected a name but doesn't have one"`.  I will tack on to the TODO list that `LowerToHW` should probably `signalPassFailure()` in this situation.

youngar, 2021-11-07:
Looks like InferWidths is not setting the width of invalid values,

Smaller test case:
```
circuit Foo:
  module Foo:
    input i: UInt<1>
    input p: UInt<1>

    wire w: UInt
    w is invalid
    when p:
      w <= i
```

```
// -----// IR Dump After InferWidths //----- //
firrtl.circuit "Foo"   {
  firrtl.module @Foo(in %i: !firrtl.uint<1>, in %p: !firrtl.uint<1>) {
    %w = firrtl.wire  : !firrtl.uint<1>
    %invalid_ui = firrtl.invalidvalue : !firrtl.uint
    firrtl.connect %w, %invalid_ui : !firrtl.uint<1>, !firrtl.uint
    firrtl.when %p  {
      firrtl.connect %w, %i : !firrtl.uint<1>, !firrtl.uint<1>
    }
  }
}
```

youngar, 2021-11-08:
One possible way to fix this: https://github.com/llvm/circt/pull/2099

youngar, 2021-11-08:
We are going to change how we are fixing this.  The problem with the canonicalization is that the return type change needs to be propagated through any operations which use the result, otherwise we fail verification implemented by the user's`InferTypeOpInterface`.  This can happen with nested `when` statements which form a chain a muxes.

The new plan is to properly infer the width of `InvalidValueOp`s inside InferWidths, instead of leaving them uninferred.  After this change is made, we won't need the canonicalizer anymore and it should be deleted.

lattner, 2021-11-20:
This works now!

closed 2021-11-20 by lattner