llvm/circt #2094: [FIRRTL/LowerToHW] Make use of unused generated random bits

lattner, 2021-11-07:
We currently generate this in the MulDiv module in RocketCore.fir:

```
      _T_18 = `RANDOM;        // Multiplier.scala:47:16
        req_fn = _T_18[3:0];    // Multiplier.scala:47:16
        _T_19 = `RANDOM;        // Multiplier.scala:47:16
        req_dw = _T_19[0];      // Multiplier.scala:47:16
 ```

The `RANDOM macro generates 32-bits of random stuff.  The second use of `RANDOM isn't necessary, it could just grab bit 4 out of _T_18.

lattner, 2021-11-07:
FPU.fir in the FIRRTL testsuite is another great example of this:

```
      _T_120 = `RANDOM;       // Reg.scala:34:16
        mem_ctrl_ren3 = _T_120[0];      // Reg.scala:34:16
        _T_121 = `RANDOM;       // Reg.scala:34:16
        mem_ctrl_swap12 = _T_121[0];    // Reg.scala:34:16
        _T_122 = `RANDOM;       // Reg.scala:34:16
        mem_ctrl_swap23 = _T_122[0];    // Reg.scala:34:16
        _T_123 = `RANDOM;       // Reg.scala:34:16
        mem_ctrl_single = _T_123[0];    // Reg.scala:34:16
        _T_124 = `RANDOM;       // Reg.scala:34:16
        mem_ctrl_fromint = _T_124[0];   // Reg.scala:34:16
        _T_125 = `RANDOM;       // Reg.scala:34:16
        mem_ctrl_toint = _T_125[0];     // Reg.scala:34:16
        _T_126 = `RANDOM;       // Reg.scala:34:16
        mem_ctrl_fastpipe = _T_126[0];  // Reg.scala:34:16
        _T_127 = `RANDOM;       // Reg.scala:34:16
        mem_ctrl_fma = _T_127[0];       // Reg.scala:34:16
        _T_128 = `RANDOM;       // Reg.scala:34:16
        mem_ctrl_div = _T_128[0];       // Reg.scala:34:16
        _T_129 = `RANDOM;       // Reg.scala:34:16
        mem_ctrl_sqrt = _T_129[0];      // Reg.scala:34:16
        _T_130 = `RANDOM;       // Reg.scala:34:16
        mem_ctrl_wflags = _T_130[0];    // Reg.scala:34:16
        _T_131 = `RANDOM;       // Reg.scala:34:16
        wb_ctrl_cmd = _T_131[4:0];      // Reg.scala:34:16
```

seldridge, 2021-11-08:
This would dramatically benefit from nailing down what random initialization actually means in FIRRTL and putting it in the spec. Does random initialization mean that one random value may be shared? Is there a single source of randomness from which we index out individual bits? Must all random sources be unique?

None of this actually matters for the purposes of generating Verilog that works with Chisel. (The random value doesn't matter, just that `x` cannot be used as the initial value due to `x`-pessimism.) The proposal here would be compatible with existing Chisel.

lattner, 2021-11-08:
I don't think this specific suggestion is that difficult.  The `RANDOM macro generates 32-bits of entropy.  There is no reason to throw them away.

I'm not suggesting that we reuse bits, just not dropping them for no reason.  Such a change would speed up the initial blocks (which doesn't matter) but more importantly will shrink the IR and generated verilog.  It is more of a "prettiness" change than anything else.

seldridge, 2021-11-08:
Ah, sure. I totally misread this. 😅  I was projecting past concerns about Verilog `x` and invalidation in FIRRTL onto this.

Completely agree. If we have 32-bits of entropy, just use them and don't throw them away.

closed 2021-11-15 by uenoku