llvm/circt #2095: [comb] missing peepholes:

lattner, 2021-11-07:
Here are some random missing peepholes seen in generated verilog from RocketCore.fir in the testsuite:

[ ] This: 

```
wire [2:0] _T_1 = {reg_debug, reg_mstatus_prv};    // reg_mstatus_prv is two bits
wire _T_57 = _T_1 > 3'h1;
```
could be: `wire _T_57 = {reg_debug, reg_mstatus_prv[1]} != 0;`

---

[ ] This: `wire [63:0] _T_0 = _T ? ~io_in2 : io_in2;`

could be strength reduced to: `wire [63:0] _T_0 = sext(_T)^io_in2;`

---

[ ] This: `io_in1[31] ? 32'hFFFFFFFF : 32'h0;` could be turned into a sext.

---

[ ] This happens a lot in the ALU module:

```
wire [31:0] _T_10 = _T_9 & 32'hFF00FF00;
wire [15:0] _T_14 = _T_10[15:0] | ...
```

could be: `wire [15:0] _T_14 = {_T_9[15:8], 8'h0}`

---

[ ] This happens a bunch in the ALU module:

```
wire [3:0] _T_58 = _T_43 & 4'hA;   // _T_58 has multiple uses.
thing = { _T_58[3:2] | _T_73[1:0], _T_58[1:0] | _T_74[3:2] }
```

This can be simplified to: `thing = { _T_43[3:2] | _T_73[1:0], _T_74[3:2] }`

seldridge, 2021-11-08:
Thanks for finding these are providing concise examples of before/after. 💯

lattner, 2021-11-29:
This is effectively done now.  The last one is actually not a noop, it is a bitswap, which I filed as Issue #2249.  There is also a broader issue with the design of SExtOp which I filed as Issue #2248

closed 2021-11-29 by lattner