llvm/circt #2107: [HW] [MSFT] [?] Commonize module/instance parsing code

teqdruid, 2021-11-09:
MSFT has its own module/module.extern/instance ops which are 90% the same as the ones in hw. Since the parsing (and printing) code is somewhat complex and lengthy, we should probably find a way to share that code (much like MLIR does with function like parsing). I don't know how much parsing code could be shared with FIRRTL.

mikeurbach, 2021-11-09:
There is already some work to pull out generic implementations here: https://github.com/llvm/circt/blob/main/include/circt/Dialect/HW/ModuleImplementation.h.

I think if we continue pushing more of HWOps.cpp into that, we should be able to refactor the MSFT version to use that implementation very smoothly. Or, maybe we could define some OpInterfaces for the different dialects to implement For modules and instances.

A couple things that the MSFT ops would like to share:

* Parameter list [printing](https://github.com/llvm/circt/blob/dc35179ce5ed350172467f2ae0e096beec7018dc/lib/Dialect/HW/HWOps.cpp#L651), [parsing](https://github.com/llvm/circt/blob/dc35179ce5ed350172467f2ae0e096beec7018dc/lib/Dialect/HW/HWOps.cpp#L519), [verification](https://github.com/llvm/circt/blob/dc35179ce5ed350172467f2ae0e096beec7018dc/lib/Dialect/HW/HWOps.cpp#L48).
* Module [printing](https://github.com/llvm/circt/blob/dc35179ce5ed350172467f2ae0e096beec7018dc/lib/Dialect/HW/HWOps.cpp#L667), [parsing](https://github.com/llvm/circt/blob/dc35179ce5ed350172467f2ae0e096beec7018dc/lib/Dialect/HW/HWOps.cpp#L545), [verification](https://github.com/llvm/circt/blob/dc35179ce5ed350172467f2ae0e096beec7018dc/lib/Dialect/HW/HWOps.cpp#L719).

mikeurbach, 2021-11-09:
For background, we had this discussion a while back: https://llvm.discourse.group/t/instancelike-and-modulelike-interfaces/4132/3.