llvm/circt #2126: [ExportVerilog] Parametric Verilog Widths Printing `w-1` as `w+MaxInt`

seldridge, 2021-11-10:
The following circuit:

```mlir
hw.module @Bar<w: i32 = 8>
  (%a: !hw.int<#hw.param.decl.ref<"w">>) -> (b: !hw.int<#hw.param.decl.ref<"w">>)  {
  hw.output %a : !hw.int<#hw.param.decl.ref<"w">>
}
```

Is printing out as `w + 0xFFFFFFFF`:

```verilog
module Bar
  #(parameter w = 8) (
  input  [w + 4294967295:0] a,
  output [w + 4294967295:0] b);

  assign b = a;	
endmodule
```

This likes like a simple problem of `-1` not showing up as intended.

seldridge, 2021-11-10:
This is actually a little tricky...

What's going on here is that a width is being converted to `#hw.param.expr.add<#hw.param.decl.ref<"w">, 0xFFFFFFFF>`.  There is a restriction, checked by an assert, that all the types of all expressions match.  I.e., everything must be signed or unsigned.  Either we need an actual `PEO::Sub` op to handle this case or we need to pattern match `+ <value with leading one>` into a subtraction.  I'm concerned that the latter approach has the potential to alias in ways that we don't want while the former is going to make expression canonicalization weirder.

Do you have any thoughts @lattner?

lattner, 2021-11-10:
I think the bug is actually a bit simpler than that :-), we should just print parameter values as signed by default.

We should also improve the printer for Add to notice when the RHS is a negative constant, and print it as minus the positive constant.  Watch out for MININT.

seldridge, 2021-11-11:
Looking into doing it this way.

There is some other funkiness going on here because the type of `#hw.param.decl.ref<"w">` isn't getting the type of the defined parameter.  🤔   Looking into this, too.

seldridge, 2021-11-11:
`#hw.param.decl.ref` is just storing a string and a type i32.  There isn't any linkage to the actual parameter on the module other than if things line up in the Verilog.  😅  It looks like this is one of those areas of the parameterization that needs to be built out more.

lattner, 2021-11-17:
I'll take this

lattner, 2021-11-18:
This patch needs to be fixed (because there is nothing special about zero).  I will take a crack at this, but @seldridge do you happen to have a specific testcase for that patch that was broken?
https://github.com/llvm/circt/commit/512620125880d09489f061e3f8b77f76bdfc15a6

The right fix for this is in firrtl -> HW lowering, not in the verilog exporter.  I will change that to use i32 parameters for its integer constants when they fit.

seldridge, 2021-11-18:
I think you're talking about https://github.com/llvm/circt/pull/2076, and not this issue?  There's an example of FIRRTL text there that was causing the problem. Any zero-valued extmodule parameter was parsed as i64 causing the emission to work this way.  I'm good with your proposal there to handle this as parsing all FIRRTL parameters into i32 (if they fit).

lattner, 2021-11-18:
Yes, that is what I meant, revised here: 75518724

lattner, 2021-11-18:
This issue is fixed in 9684a353, thanks!

closed 2021-11-18 by lattner