llvm/circt #2130: [FIRRTL] Preserve vectors in lowering

uenoku, 2021-11-10:
Currently, LowerTypes flattens all aggregated types. Therefore, this firrrtl: 
```firrtl
circuit Foo:
  module Foo:
    input clock: Clock
    input a: UInt<1>[2][2]
    output b: UInt<1>[2][2]

    reg r: UInt<1>[2][2], clock
    r <= a
    b <= r
```
is converted into 
```verilog
module Foo(     // foo.fir:2:10
  input  clock, a_0_0, a_0_1, a_1_0, a_1_1,
  output b_0_0, b_0_1, b_1_0, b_1_1);

  reg r_0_0;    // foo.fir:7:5
  reg r_0_1;    // foo.fir:7:5
  reg r_1_0;    // foo.fir:7:5
  reg r_1_1;    // foo.fir:7:5

  always @(posedge clock) begin // foo.fir:8:7
    r_0_0 <= a_0_0;     // foo.fir:8:7
    r_0_1 <= a_0_1;     // foo.fir:8:7
    r_1_0 <= a_1_0;     // foo.fir:8:7
    r_1_1 <= a_1_1;     // foo.fir:8:7
  end // always @(posedge)

  assign b_0_0 = r_0_0; // foo.fir:2:10
  assign b_0_1 = r_0_1; // foo.fir:2:10
  assign b_1_0 = r_1_0; // foo.fir:2:10
  assign b_1_1 = r_1_1; // foo.fir:2:10
endmodule
```
but preferably the following is better in terms of readability and simulation costs:
```verilog
module Foo(     // foo.fir:2:10
  input             clock,
  input  [1:0][1:0] a,
  output [1:0][1:0] b);

  reg [1:0][1:0] r;     // foo.fir:7:5

  always @(posedge clock)       // foo.fir:8:7
    r <= a;     // foo.fir:8:7
  assign b = r; // foo.fir:2:10
endmodule
```
It is better for bundles to be lowered into `sv.interface` but in the issue, I'd like to limit the scope to preserve vectors. 
I think base vector types(N-dim array not containing bundle) can be lowered directly.

For this, there are following tasks across several passes:

- [ ] IMConstantProp <br>It must be field sensitive

- [ ] LowerToHW <br> Initialize random value for arrays

- [ ] LowerTypes <br> Change not to flatten base vector types 

Maybe outdated but related to:
* https://github.com/llvm/circt/issues/123
* https://llvm.discourse.group/t/firrtl-aggregate-types-discussion/1806/23

uenoku, 2021-11-10:
I'd like to hear other concerns missing here!

seldridge, 2021-11-10:
I _think_ that these should show up as _unpacked_ vector dimensions as opposed to _packed_ dimensions.  However, this shouldn't change the actual implementation of this much either way.

lattner, 2021-11-10:
Awesome, thank you for looking into this.  From a procedural perspective, I think that it would make sense to work on moving each of the passes which is currently "after legalizetypes" to run "before legalize types" one at a time.  This ensures that we're hill climbing.    Each of those passes would need extensions to work (e.g. being field sensitive)

> I think that these should show up as unpacked vector dimensions as opposed to packed dimensions. However, this shouldn't change the actual implementation of this much either way.

I think the examples are correct here: we're looking for these to be treated as blocks of aggregate bits, we don't want mem inference or other wacky things to kick in.

seldridge, 2021-11-10:
> Awesome, thank you for looking into this. From a procedural perspective, I think that it would make sense to work on moving each of the passes which is currently "after legalizetypes" to run "before [...]

👍 

> I think the examples are correct here: we're looking for these to be treated as blocks of aggregate bits, we don't want mem inference or other wacky things to kick in.

Thinking about this more, I'm convinced.  Chisel and FIRRTL have _separate_ constructs for memories.  If they use vectors, they are implying packed dimensions.  This does imply that there is a semantic difference between a FIRRTL memory and a register with a vector type.  I think that is actually quite clean, though. 👍

uenoku, 2021-12-03:
ExpandWhens support vectors by https://github.com/llvm/circt/pull/2286. And the next step would change LowerTypes not to lower aggregates into ground type wire/reg. Instead, "ExpandConnet" (i.e. making all connections are between ground types) is necessary. I would like to create a flag to switch behavior so that I don't break anything.  

As an implementation, I believe ExpandConnect and LowerTypes are not much different.  I think I can implement ExpandConnect in LowerTypes but  creating a new pass might be better.

youngar, 2021-12-03:
It might be okay to handle ExpandConnects in ExpandWhens, my thinking is:
1. The logic to expand a connect is not too difficult, it shouldn't add too much complexity to the pass
2. ExpandWhens creates a connection for registers to itself, for an aggregate register it will have to do expand a bulk connect here anyways
3. If we ever want to try to keep some bulk connects around, it probably makes sense to put all the logic inside expand whens anyways
WDYT?

uenoku, 2021-12-03:
Thanks for the comment!

> It might be okay to handle ExpandConnects in ExpandWhens, my thinking is:

It would work but my concern is that we need to lower passive types anyway in somewhere. Currently, LowerTypes is doing it very wisely and the logic of ExpandConnect will be very similar to LowerTypes. So that's why I want to place expand connections close to LowerTypes.

> ExpandWhens creates a connection for registers to themself, for an aggregate register it will have to expand a bulk connect here anyways

Yes, I agree. We need to do ExpandConnect there. 

> If we ever want to try to keep some bulk connects around, it probably makes sense to put all the logic inside expand whens anyways

Yeah, makes sense. What I'm thinking about ExpandConnect is that we don't have to expand all the connections actually. My idea is to find "the most precise subelments in the module" and only expand around them, for example:

```scala
a: UInt<1>[50][50]
reg b: UInt<1>[50][50]
a <= b
a[0] <= b[0]
a[2] <= b[2]
a[2][2] <= b[2][2]
```
In this case, there exists subindex op `a[0]`, therefore `a` must be expanded. And next, there exists `a[2][2]` so `a[2]` must be expanded. As a result, we don't have to expand most of their elements such as `a[1], a[3], a[4], ...   a[50]`, which would be nice.

uenoku, 2021-12-06:
Talked to @darthscsi and it seems that modifying LowerTypes not to lower passive types would be good way to go.