llvm/circt #2131: IMConstantProp It must be field sensitive

uenoku, 2021-11-10:


uenoku, 2021-11-14:
I thought about the extension to represent aggregate types in lattice. I think (only?) one option would be to use inheritance and virtual functions like this:

```cpp
class LatticeValue {
  ....
  virtual void markOverdefined() = 0;
  virtual void markInvalidValue(InvalidValueAttr value = 0;
  virtual void markConstant(IntegerAttr) = 0;
  virtual bool isUnknown() const = 0;
  virtual bool isInvalidValue() const = 0;
  virtual bool isConstant() const = 0;
  virtual bool isOverdefined() const = 0;
}

class GroundTypeLatticeValue : public LatticeValue {
  ...
  llvm::PointerIntPair<Attribute, 2, LatticeValue::Kind> valueAndTag;
}

// Use for bundle and vector types
class AggregateTypeLatticeValue : public LatticeValue {
  ....
  // index to lattice value
  SmallVector<std::unique_ptr<LatticeValue>> values;
};
```
This change might cause performance regressions so do you have any thoughts about this?

seldridge, 2021-11-14:
I'm not entirely convinced of the best approach here.  One thing to consider is that aggregate lattice values are really (as you show) just storage for the underlying ground type lattice values.  It may be possible to keep the lattice values as ground types and then do more complicated updating of these when a connect/partial connect involves a ground type.  E.g., when examining an aggregate connect, you need to figure out which lattice values on the LHS (and RHS if dealing with bundles with flips on the LHS) need to get updated.

One alternative would be to do an "expand connects" pass like the Scala FIRRTL Compiler does.  This converts all aggregate connections to connections between ground types.  After this, the existing IMCP pass may be a lot simpler to update because it shouldn't see any aggregate types on the operations its sensitive to, i.e., connects and partial connects.  Doing it this way would, however, mean connects are blown out and we may not want to lower out of aggregate connections.

uenoku, 2021-11-14:
> One alternative would be to do an "expand connects" pass like the Scala FIRRTL Compiler does. This converts all aggregate connections to connections between ground types. After this, the existing [...]

I see, it sounds exactly what `LowerTypes` is currently doing for vector types 😅

lattner, 2021-11-14:
It is possible to expand out the lattice value so that you maintain the "one lattice value to one SSA value" invariant we have now, and then model each field as part of that huge aggregate lattice value.  The problem with this is that the "height" of the lattice will be very high, and propagation through the graph will cause many many many lattice transitions as each field gets resolved.  This will lead to inefficient compilation.

The alternative approach is to change LatticeValues from `DenseMap<Value, LatticeValue> latticeValues;` to `DenseMap<pair<Value, unsigned>, LatticeValue> latticeValues;`, where the unsigned value is the field # in question (the depth first enumeration of the field position).  This allows each of the lattice values to be tracked independently, and allows updates to only flow through to the users that are affected.

This only works in cases where we have static knowledge of which field is being accessed: this is true for struct fields but not for arrays that have subaccesses with dynamic index.  There are a couple different ways to handle that, but it will take some exploration.

uenoku, 2021-11-24:
> One alternative would be to do an "expand connects" pass like the Scala FIRRTL Compiler does. This converts all aggregate connections to connections between ground types. After this, the existing [...]

Oh, I have misunderstood "expand connects". It makes a lot sense. Considering that we have to modify ExpandWhen too, I feel it is much better representation.

dtzSiFive, 2023-06-05:
Handled by #4795 .

closed 2023-06-05 by dtzSiFive