llvm/circt #2172: [Website] circt.llvm.org dialect list has two `hw` dialects listed

teqdruid, 2021-11-12:
[image]

I don't know how this is generated, but one `hw` is actually `hw`. The other is `sv`.

youngar, 2021-11-12:
Just took a quick look at this, it looks like the SV.md doc generated by tablegen includes both the HW and SV dialects.  You can kinda see this here in the online page: https://circt.llvm.org/docs/Dialects/SV/#a-uniontype

Looking at the [tablegen code](https://github.com/llvm/llvm-project/blob/main/mlir/tools/mlir-tblgen/OpDocGen.cpp#L376) for this, it looks like it produces docs for all dialects defined in the tablegen file its pointed at. I'm guessing this is probably a side effect of some of dialect td files including each other.

I don't think this is really in @drom's wheelhouse, so I'm going to unassign it from him.

teqdruid, 2021-11-12:
Thanks for the initial investigation @youngar !

maerhart, 2021-11-13:
I think this is because of including `HWAttributesNaming.td` [here](https://github.com/llvm/circt/blob/main/include/circt/Dialect/SV/SV.td#L40) which contains an `AttrDef`.

A solution could be to not include that directly in SV, but create a `DialectAttr` referring to the `AttrDef` of `InnerRefAttr` and just include that in SV (as also done in `HWAttributes.td` for the implementations and `HWTypes.td` for the includable predicates).

I observed the same/similar issue while adopting HW types in LLHD.

youngar, 2022-05-25:
This is fixed by https://github.com/llvm/llvm-project/commit/e0c3b94c80143376473ec7110ca0c8a4fe03112e, thanks @River707!  cc @mkorbel1

closed 2022-05-25 by youngar