llvm/circt #2175: [LowerToHW?] Generating slice syntax for no reason

lattner, 2021-11-15:
I'm seeing a lot of diffs like this, we should go back to the old form.  I don't see where in the codebase we are creating ArraySliceOp's though!

```
   reg [27:0] Memory[0:31];
 
   always_ff @(posedge RW0_clk) begin
     if (RW0_en & RW0_wmask & RW0_wmode)
-      Memory[RW0_addr] <= RW0_wdata;
+      Memory[RW0_addr][32'h0 +: 28] <= RW0_wdata;
   end // always_ff @(posedge)
```

We should also canonicalize away ArraySliceOp's of their entire contents.

lattner, 2021-11-15:
Ah, these are IndexedPartSelect's!

lattner, 2021-11-15:
@prithayan can you take a look at this, this looks like a regression form your recent patch.

lattner, 2021-11-15:
One way to solve this would be to make `IndexedPartSelectInOutOp` have a folder that nukes noop selects, and change HWMemSimImpl to use createOrFold instead of `create<IndexedPartSelectInOutOp>`

prithayan, 2021-11-15:
Thanks Chris, this was next on my TODO list. The HWMemSimImpl is creating these ops unnecessarily. Will add the folders.

closed 2021-11-15 by prithayan