llvm/circt #2191: [StaticLogic] Additional attributes to pipeline operation.

mikeurbach, 2021-11-16:
This came up in #2183, but more generally, I think there are a couple things we could do to improve the pipeline represenation. Essentially, this boils down to capturing more information, which is readily available from the scheduling problem, or would need to be supplied from elsewhere.

#### Stage start times

We currently model the stages in a "dense" way. I.e., we implicitly model the start time of each stage by its position in the list of stages. For example, if there is a multi-cycle operation that can't be used from some stages, we would need to introduce empty stages in the current representation, like so:

```mlir
%1 = staticlogic.pipeline.stage {
  %0 = multi_cycle(...)
  staticlogic.pipeline.register %0
}
staticlogic.pipeline.stage {}
staticlogic.pipeline.stage {}
staticlogic.pipeline.stage {
  use(%1)
}
```

This is unnecessary, and instead I propose to use a "sparse" storage, based on start times, like so:

```mlir
%1 = staticlogic.pipeline.stage startTime = 0 {
  %0 = multi_cycle(...)
  staticlogic.pipeline.register %0
}
staticlogic.pipeline.stage startTime = 3 {
  use(%1)
}
```

Julian also brought up that you can model predicated execution in a neat way, i.e.:

```mlir
staticlogic.pipeline.stage startTime=0 when(%0) {
  ...
}
staticlogic.pipeline.stage startTime=0 when(%1) {
  ...
}
```

I propose to include such an attribute on the stage operator to remove the need to have empty stages, and support the predicated use-case.

#### Operator latencies and initiation intervals

Another thing that is currently explicit when the pipeline is constructed, but implicit in the representation, is the latency of operators. It is the job of the pass or tool creating the pipeline to ensure results are used in the appropriate stage.

Another point that we haven't really discussed, but came up in #2183, is some operators may be pipelined themselves. Their latency could be greater than their initiation interval. Being able to exploit "nested" pipelinining could be important.

Do we need to model operator latencies and IIs in the pipeline? What does this aid, besides providing a way to add more verifications?

If we do decide we want to track this, I don't have a great proposal besides slapping down discardable attributes on arbitrary ops, i.e.:

```mlir
%1 = staticlogic.pipeline.stage startTime = 0 {
  %0 = arith.muli ..., ... { latency=3, II=1 }
  staticlogic.pipeline.register %0
}
```

We could verify the use of %1 is in a stage of `startTime + latency`, and if used as an iter_arg, that `II` is respected.

stephenneuendorffer, 2021-11-16:
Not to discourage you, but I think this modeling predicates and start times is going to make alot of the calculus about pipelines more complicated.  I would suggest keeping a simple representation to start and to determine what are the analysis interfaces that make sense to expose to algorithms.  I think we need to stand up basic algorithms first and then come back to see how these representational choices might make designs easier to capture or algorithms more efficient, in order to avoid too much premature optimization.

I would propose that if you're going to represent latency on operators, then a `use()` can appear any stage *after* the latency, rather than exactly in the right place: it's valid to insert additional pipeline stages, but the latency is a minimum.

mikeurbach, 2021-11-16:
Thanks for the feedback. I do want to avoid baking too much into this operation before we have a meaningful reason to. I'm starting to transform the representation as it is defined today into Calyx, and I'm happy to let that drive any future enhancements.

The proposal to add a start time came up because in #2183  we wanted to verify the II of a pipeline was legal, and it seemed a startTime attribute would make that more straightforward. I think it can be done based on the current representation, so again I'm happy to table this discussion till there is a more pressing need.

mikeurbach, 2021-11-18:
Closing this, but opening a new thread of discussion here: https://github.com/llvm/circt/issues/2204.

closed 2021-11-18 by mikeurbach