llvm/circt #2195: MLIR merge would be helpful

lattner, 2021-11-16:
There are a bunch of [compile time speedups](https://reviews.llvm.org/D113511) and [other improvements](https://reviews.llvm.org/D113641) in mainline MLIR, it would be great to merge these into the CIRCT tree.

seldridge, 2021-11-16:
I looked at this briefly over the weekend and there are some things that will need updating:

- [`mlir::Identifier` was removed in favor of using `mlir::StringAttr`](https://reviews.llvm.org/D113536)
- A possible change to `parse` and `print` function signatures (which I couldn't trace to an upstream commit)

lattner, 2021-11-17:
The `m_Any` definition and related class can go away when this is done, as commented here: d67c9c93

uenoku, 2021-11-20:
> A possible change to parse and print function signatures (which I couldn't trace to an upstream commit and may have been due to me botching the update)

It looks we have to use AsmParser/AsmPrinter in operation implementation (https://reviews.llvm.org/D113637)

closed 2021-12-01 by seldridge