llvm/circt #2203: [FIRRTL] Cast Folders for Invalid

seldridge, 2021-11-18:
Similar to #2197, there are other folds/canonicalizations that are missing that may result in registers hanging around.

Consider the following.  This is the same as #2197, but just with `asUInt` as opposed to `pad`:

```scala
circuit Foo:
  module Foo:
    input clock: Clock
    output a: UInt<1>

    wire x: SInt<1>
    x is invalid

    reg b: UInt<1>, clock
    b <= asUInt(x)
    a <= b
```

MFC produces a register:

```verilog
  always @(posedge clock)	// Foo.fir:10:7
    b <= 1'h0;	// Foo.fir:7:7, :10:7
  assign a = b;	// Foo.fir:2:10
```

SFC produces a constant output:

```verilog
  assign a = 1'h0;
```

fabianschuiki, 2021-11-18:
I'm wondering if this is "just" a matter of getting the unary cast-like operations in. Or should we, for the sake of canonicalization, generally treat `InvalidValueOp` equivalent to a constant zero in folders/canonicalizers? For example, `add(x, y)` with `y` being another `invalidvalue` should probably still be folded away.

Or we could introduce a form of "invalid poisoning" where an invalid operand on most unary/binary ops will make the entire op fold to an invalid. With a similar reasoning as X propagation in SV/VHDL, where an invalid/unknown value at an input causes an undefined output. The problem here being that the Scala FIRRTL compiler treats `invalidvalue` as a 0, so `add(invalid, x)` would generally fold to `x`, while my suggestion would fold to `invalid` :woozy_face:

seldridge, 2021-11-18:
>  The problem here being that the Scala FIRRTL compiler treats invalidvalue as a 0, so add(invalid, x) would generally fold to x, while my suggestion would fold to invalid 🥴

There's two "flavors" of invalid and MFC gets into trouble when it conflates them.  SFC can do this because it doesn't actually have an invalid value and it runs constant propagation very late in the pipeline.  SFC treats invalid removal as the problem of two separate passes (expand whens and invalid to zero conversion) and then constant propagation runs never seeing a true invalid value.

This is why I'm leaning towards canonicalize away all invalid values.  I think we can treat `mux(en, invalid, x)` as one canonicalization and then everything else is a canonicalization treating invalid as constant zero.

Invalid poisoning probably makes sense for a Chisel API, but I get spooked trying to overlay that onto invalid. 😬 

Two views of invalid:

### Default in When Statement

```scala
wire x: UInt<1>
x is invalid
when en:
  x <= UInt<1>(1)
```

The SFC converts this to `validif` (which we removed in favor of `mux(en, value, invalid)`.

```scala
wire x: UInt<1>
x <= validif(en, UInt<1>(1))
```

The SFC then constant props this to:

```scala
wire x: UInt<1>
x <= UInt<1>(1)
```

### Constant Zero

```scala
wire x: UInt<1>
x is invalid
node y = add(x, UInt<1>(1))
```

SFC converts this to constant zero:
```scala
wire x: UInt<1>
x <= UInt<1>(0)
node y = add(UInt<1>(0), UInt<1>(1))
```

fabianschuiki, 2021-11-18:
This is a nice writeup of the problem! Looking at it this way I totally agree that we might just canonicalize away all the `invalid`s as soon as possible. What was the reason we preserved that `validif`/`mux(en, value, invalid)` again? Couldn't we just immediately fold that away in `ExpandWhens`?

fabianschuiki, 2021-11-18:
Cool, that sounds like it might be a good thing to do. Not sure how we would implement this. Because some of the canonicalizers, like mux, would like to see the invalidvalue and fold accordingly. Others want to see it as a zero already, e.g. `add(x, invalid)`. I suspect that we might want to extend the `IsConstantZero` (or similar) predicate and the constant value matcher, such that most patterns will mach `invalidvalue` as a zero constant, except for a few folds that can opt out of that behavior because they know something better to do with `invalidvalue` (like muxes).

seldridge, 2021-12-02:
I did some more digging and there's further _different_ behavior for invalid involving registers (which you originally added in the SFCTests `remove-reset.fir`). This requires preserving the connections from an invalid value to a register initialization and using this information to _NOT_ give the register a reset.  

This results in puzzles like the following.  Here invalid _should_ be propagated to the register initialization, but _should not_ be propagated to the default condition in the when.

```scala
circuit Foo :
  module Foo :
    input clock: Clock
    input reset: UInt<1>
    input cond: UInt<1>
    input d: UInt<1>
    output q: UInt<1>
    output x: UInt<8>

    ; This is an invalid wire.  This is the closest thing FIRRTL and the SFC
    ; have to an invalid value.
    wire inv: UInt<1>
    inv is invalid

    ; This register is initialized to an invalid value.  This results in the
    ; register NOT getting a reset in the output Verilog.
    reg r: UInt<1>, clock with : (reset => (reset, inv))
    r <= d
    q <= r

    ; This is a structure describing an invalidated output.  This should result
    ; in a mux with zero being generated.
    x <= inv
    when cond:
      x <= UInt<8>(42)
```

I think what we need to do is:

1. Add all the canonicalizations to primops to treat invalid values like they were zero (like you are doing with https://github.com/llvm/circt/pull/2251)
2. Canonicalize invalid values that hit a mux leg _directly_ without any intermediary connection.
3. Not rely on IMCP/Canonicalization to do the RemoveResets analysis.

Fundamentally, I don't think invalid value can ever be propagated without destroying information. 🥲

lattner, 2021-12-02:
Is there a reason we _need_ to propagate them?  Why not just be super conservative like SFC is?

Are we trying to "optimize" things more than is required?

seldridge, 2021-12-02:
> Is there a reason we need to propagate them? Why not just be super conservative like SFC is?

I think that makes sense.

We currently propagate them to remove initializations.  However, we don't have to do that with canonicalization/folding and can instead use the same algorithm the SFC uses (look backwards through connects and see if you hit an invalid).

> Are we trying to "optimize" things more than is required?

Likely. Particularly around invalid.

lattner, 2021-12-02:
Last time we did that (with the flip representation) we got burned.  We should save cleverness for HW level IMO, and go for full compat at the firrtl dialect level

seldridge, 2022-07-03:
This should be mostly fixed on `main` through the use of the `SFCCompat` pass which resolves invalid interpretations that should not be lowered to zero. There are lit tests which check for this behavior now.

closed 2022-07-03 by seldridge