llvm/circt #2216: exportverilog crash

lattner, 2021-11-19:
@uenoku points out that this crashes:

```
$ cat x.mlir
hw.module @EmittedDespiteDisallowed(%clock: i1, %reset: i1) {
  %tick_value_2 = sv.reg  : !hw.inout<i1>
  %counter_value = sv.reg  : !hw.inout<i1>
  sv.initial {
    %RANDOM = sv.verbatim.expr.se "magic" : () -> i2 {symbols = []}
    %1 = comb.extract %RANDOM from 0 : (i2) -> i1
    %2 = comb.extract %RANDOM from 1 : (i2) -> i1
    sv.bpassign %tick_value_2, %1 : i1
    sv.bpassign %counter_value, %2 : i1
  }
  hw.output
}
$ circt-opt --lowering-options=disallowLocalVariables --export-verilog x.mlir
```

I'll take a look when I get a chance.

uenoku, 2021-11-19:
Thanks! Actually, @fabianschuiki points out this with circt-reduce :)

uenoku, 2021-11-19:
I took a look at this and I think the following condition has to care about whether or not op creates temporary wires. https://github.com/llvm/circt/blob/e37980d162663db319fcbb8021e3165aecce38f5/lib/Conversion/ExportVerilog/PrepareForEmission.cpp#L442

fabianschuiki, 2021-11-19:
Excellent find @uenoku! I had a really hard time finding this piece of code on my own 🙈. Maybe this is missing some "is used twice" part in the condition to make it spill to a wire as well?

uenoku, 2021-11-19:
`PrepareForEmission.cpp` is currently doing local variable hoisting. I think it the result of `comb.extract` must be hoisted to the parent region.  
```mlir
hw.module @EmittedDespiteDisallowed(%clock: i1, %reset: i1) {
  %tick_value_2 = sv.reg  : !hw.inout<i1>
  %counter_value = sv.reg  : !hw.inout<i1>
  sv.initial  {
    %magic = sv.verbatim.expr.se "magic" : () -> i2 {symbols = []}
    %0 = comb.extract %magic from 0 : (i2) -> i1
    sv.bpassign %tick_value_2, %0 : i1
    sv.bpassign %counter_value, %0 : i1
  }
  hw.output
}
```
===> After `prepareHWModule`
```mlir
hw.module @EmittedDespiteDisallowed(%clock: i1, %reset: i1) {
  %tick_value_2 = sv.reg  : !hw.inout<i1>
  %counter_value = sv.reg  : !hw.inout<i1>
  %0 = sv.reg  : !hw.inout<i2>
  sv.initial  {
    %magic = sv.verbatim.expr.se "magic" : () -> i2 {symbols = []}
    %1 = sv.read_inout %0 : !hw.inout<i2>
    sv.bpassign %0, %magic : i2
    %2 = comb.extract %1 from 0 : (i2) -> i1 // %2 is not hoisted
    sv.bpassign %tick_value_2, %2 : i1
    sv.bpassign %counter_value, %2 : i1
  }
  hw.output
}
```

I think it should be converted into:
```
hw.module @EmittedDespiteDisallowed(%clock: i1, %reset: i1) {
  %tick_value_2 = sv.reg  : !hw.inout<i1>
  %counter_value = sv.reg  : !hw.inout<i1>
  %0 = sv.reg  : !hw.inout<i2>
  %extract = sv.reg  : !hw.inout<i1> // It is needed
  sv.initial  {
    %magic = sv.verbatim.expr.se "magic" : () -> i2 {symbols = []}
    %1 = sv.read_inout %0 : !hw.inout<i2>
    sv.bpassign %0, %magic : i2
    %ext_in_out = sv.read_inout %extract : !hw.inout<i1>
    %ext_tmp = comb.extract %1 from 0 : (i2) -> i1
    sv.bpassign %extract, %ext_tmp : i1
    sv.bpassign %tick_value_2, %ext_in_out : i1
    sv.bpassign %counter_value, %ext_in_out : i1
  }
  hw.output
}
```

uenoku, 2021-11-19:
>  Maybe this is missing some "is used twice" part in the condition to make it spill to a wire as well?

As far as I see it, the cause seems that we are missing the handling when `hoistNonSideEffectExpr` fails.
https://github.com/llvm/circt/blob/47a4cd20631cecf7b655120003aa3b0e75cebf9b/lib/Conversion/ExportVerilog/PrepareForEmission.cpp#L454-L463
```

```

fabianschuiki, 2021-11-19:
Oohh that's a clever observation. I wasn't aware we were doing this side-effect related hoisting in the code. That would also explain why the verbatim in this case wouldn't be moved (rightfully so).

lattner, 2021-11-19:
Oh yeah, I wouldn't be surprised at all if some transformations in "prepare" need to be gated/disabled by this flag.

lattner, 2021-11-19:
This looks like a bug in rewriteSideEffectingExpr, working on it.

lattner, 2021-11-19:
Fixed in 6c921321, please give that a try.

closed 2021-11-19 by lattner

uenoku, 2021-11-19:
Oh, thank you for the fast fix! It's like I was looking in the wrong place 😅