llvm/circt #2219: [HW] Reconsider zero width wires

lattner, 2021-11-19:
Now that APInt supports zero width values, we can make hw.constant support them, and expand comb to allow them.

Such an approach would make it easier to generate HW dialect, make some transformations easier, and would define away corner cases (e.g. the array index of size=1 needing a zero width size).  OTOH, it will make verilog export more complicated.

If someone is interested in this, we should discuss at a design meeting.

maerhart, 2021-11-24:
Zero width values would be very helpful from a Moore/LLHD perspective as the Moore frontend frequently emits `array_get` operations on size=1 arrays, but also other extract/slice operations requiring `i0` indices.

teqdruid, 2022-09-19:
Did we ever discuss this at an ODM? I don't recall discussing it and I can't find an entry in the meeting notes.

I've seen a lot of edge cases which needed to be special cased due to this lack of support, so I'm strongly in favor.

mortbopet, 2022-09-29:
As of #3985, i0 is now legal for `HWIntegerType`'d values.

closed 2022-09-29 by mortbopet