llvm/circt #2224: [ExportVerilog] Symbols defined in `IfDefOp` regions are not added to symbol cache

teqdruid, 2021-11-20:
Symbols get added to the symbol cache in `gatherFiles` ([specifically this `TypeSwitch`](https://github.com/llvm/circt/blob/d2e29b53634ca64d66c1d32f57210f4ec40d0fe5/lib/Conversion/ExportVerilog/ExportVerilog.cpp#L4073)). The typeswitch doesn't include logic for `IfDefOp`, which must be recursive. It's not only difficult to make this function recursive it's not clear that it makes sense to let operations not at the top level get emitted to different files.

I think populating the symbol cache should be divorced from gathering the files. (The function documentation for `gatherFiles` doesn't mention populating the symbol cache anyway.) @lattner what do you think?

For reference, I came across this bug in an effort to put `TypeScopeOp` in an `IfDefOp` as a c-style header guard, so really any solution that results in `TypeScopeOp` symbols inside of `IfDefOp`s making it into the symbol cache satisfies my immediate needs. This problem seemed better than that.

lattner, 2021-11-23:
I'm falling behind on the type scope stuff (I'll look at it when I get a chance), but there are bigger questions here and I'd prefer we move slowly.  There are a bunch of inter-related issues here, and I'm concerned that fixing individual things is going to create a bit of a mess.

It is pretty trivial to make gatherFiles pick up symbols recursively through ifdefs, but that won't be enough and will turn into a different set of problems:
 - The splitting across files cannot be recursive, it can only be defined on top level entities as you say.
 - it would be /really/ bad to try to define the same symbol to different things on different arms of an ifdef.  
 - Instances in ifdefs within modules have a similar issue - we definitely don't support them reliably.

Can you share more about what you're trying to accomplish here at the design meeting next week?  The type scope stuff is also not really working, and I'd love to understand what you're trying to get done, because there may be alternate solutions.

teqdruid, 2021-11-23:
> Can you share more about what you're trying to accomplish here at the design meeting next week?

Sure -- I've added it to the meeting notes. It's actually pretty simple: I need to put the typedefs in a header guard:

```
`ifdef __TYPES__
`define __TYPES__
typedef i8 foo;
`endif
```

PyCDE has moved to a model wherein we split the output SV into a few different files -- opening a single very long file was getting cumbersome (and exposing performance issues with all of the vscode systemverilog extensions). Right now, we include the typedefs at the beginning of each file. If, however, all the files are in the same translation unit in certain situations Questa complains about typedef redefinitions. So we eventually want to get to separating out the typedefs into a header file and including it at the top of each file. This is standard in hand-written systemverilog.

Since we can rely on VerbatimOp ordering at the top level, I'll just use that for the ifdef/endif. I also considered adding a "guard" attribute to typescopes so that they'd automatically output header guards is specified.

Why do we even have IfDefOp if it's not well supported? Is anyone successfully using it?

> The type scope stuff is also not really working, and I'd love to understand what you're trying to get done, because there may be alternate solutions

Yeah, we said we'd be revisiting this at some point in the future since the existing solution is a hack.

> It is pretty trivial to make gatherFiles pick up symbols recursively through ifdefs, but that won't be enough and will turn into a different set of problems

Is there a particular reason that `gatherFiles` does double duty in picking up symbols? They seem like two different tasks and look pretty seperable.

mikeurbach, 2021-11-23:
> >  The type scope stuff is also not really working, and I'd love to understand what you're trying to get done, because there may be alternate solutions
>
> Yeah, we said we'd be revisiting this at some point in the future since the existing solution is a hack.

Are you guys referring to this issue? https://github.com/llvm/circt/issues/1642. Or some other aspect of the current design that is busted?

teqdruid, 2021-11-23:
> Are you guys referring to this issue? #1642. Or some other aspect of the current design that is busted?

Yeah, that's the issue to which I was referring.

lattner, 2021-12-02:
Per discussion today, I think we can close this.  Could you file another issue to talk about the `include` changes we discussed in the design meeting today?