llvm/circt #2229: what is the output format of llhd-sim？

hunterzju, 2021-11-23:
Hi, @fabianschuiki , I tried the `llhd-sim` in the project and the output format seems is not the `vcd` format? Is there a tool to convert to the vcd format so that I can use gtkwave ? 
And I want to know what is the future of the LLHD dialect, will it go further or merge into the FIRRTL dialect? Thanks!

fabianschuiki, 2021-11-23:
Hi @hunterzju! `llhd-sim` currently produces its own trace format which is very easy to debug -- it's basically just a list of all the events and changes that occur. You can control how much of that you'd like to see with the `--trace-format` option (see `llhd-sim -h`). We definitely plan to add VCD output (and others) at a later point, just haven't had the time yet. There might be a script somewhere that converts from the trace format to VCD; @maerhart or @rodonisi might know better.

My plan is to have the LLHD dialect remain there as a representation of the event-based simulation semantics of SV and VHDL. It is a lower level of abstraction than FIRRTL and can describe more things. Ideally LLHD is there to tie in SV/VHDL source code and provide a fallback dialect for simulation (because event-based always works, but it is likely a lot slower than simulating HW/Comb/Seq directly, whose semantics are more streamlined).

maerhart, 2021-11-23:
I only remember a python script to convert VCD files to the trace format `llhd-sim` emits that @rodonisi wrote.

hunterzju, 2021-11-24:
Thanks for reply! I am not sure the meaning of `delta` and `eps` in `struct Time` in the file `state.h`.

for example, I have the verilog code like this :
``` verilog
 initial begin
 clk=0;
 forever #5ns clk=~clk;
 end
```
and the mlir file generated from moore looks like this:
```mlir
llhd.proc @updowncounter_tb.initial.63.0() -> (%clk: !llhd.sig<i1> ) {
     br ^0
 ^0:
     %1 = llhd.prb %clk : !llhd.sig<i1>
     %clk.shadow = llhd.var %1 : i1
     %2 = hw.constant 0 : i1
     %3 = llhd.constant_time #llhd.time<0s, 0d, 1e>
     llhd.drv %clk, %2 after %3 : !llhd.sig<i1>
     llhd.store %clk.shadow, %2 : !llhd.ptr<i1>
     br ^loop_body
 ^loop_body:
     %4 = llhd.constant_time #llhd.time<5ns, 0d, 0e>
     llhd.wait  for %4, ^5
 ^5:
     %6 = llhd.prb %clk : !llhd.sig<i1>
     llhd.store %clk.shadow, %6 : !llhd.ptr<i1>
     %clk.shadow.ld = llhd.load %clk.shadow : !llhd.ptr<i1>
     %7 = hw.constant -1 : i1
     %8 = comb.xor %7, %clk.shadow.ld : i1
     %9 = llhd.constant_time #llhd.time<0s, 0d, 1e>
     llhd.drv %clk, %8 after %9 : !llhd.sig<i1>
     llhd.store %clk.shadow, %8 : !llhd.ptr<i1>
     br ^loop_body
 }
```
The `%3` and `%9` is a llhd.time value with `eps=1`, what is the meaning of the eps value?