llvm/circt #2253: [FIRRTL] Failing nightly integration test

teqdruid, 2021-11-29:
[image]

Not sure why [this is failing](https://github.com/llvm/circt/actions/runs/1510446247) only in the nightly. Been failing for a [few nights](https://github.com/llvm/circt/actions?query=workflow%3A%22Nightly+integration+tests%22).

lattner, 2021-11-29:
This is due to @fabianschuiki 's patch here: f08727b7bb36d7a702b20c944e909db048ecc4ea

Fabian, can you take a look and/or revert?

seldridge, 2021-11-29:
There's two solutions to fix this: 
1. Remove the assertion (because it's unneeded as no IR mutation takes place when the assertion is checked)
2. Split emission into parallel and sequential regions: https://github.com/llvm/circt/pull/2238

There is more discussion on these two approaches on #2238.

fabianschuiki, 2021-11-30:
Will be fixed by #2256.

closed 2021-11-30 by fabianschuiki