llvm/circt #2254: firrtl DontTouchAnnotation "unapplied annotation with target"

lsteveol, 2021-11-29:
Forgive me if this question has been answered, is documented, or is fairly straightforward, I'm just starting with CIRCT experimentation (I am also a complete LLVM/MLIR n00b). I have a Chisel design in which I'm using `firtool` to generate the Verilog. When I add an annotation file, I begin to get several errors on pretty much all of my annoations (`unapplied annotation with target`). 

```
/home/lsteveol/projects/wlink_test/wav-wlink-hw/wav.wlink.AXI128bit6LaneWlinkTestConfig/WlinkSimpleTestHarness.fir:2:1: error: unapplied annotations with target '~WlinkSimpleTestHarness|WlinkGPIOPHY>scan_out' and payload '[{class = "firrtl.transforms.DontTouchAnnotation"}]'
```
I'm using `firtool` for the Verilog Generation, and also using the `--annotation-file=<>` flag, thinking that would allow me to pass these annotations, but I keep running into these issues. On the Chisel weekly several weeks back, I recall there being some issues with Annotations (primarily `BoringUtils` if my memory serves me correctly). 

Here is what I'm running with `firtool`
```
/home/lsteveol/projects/circt-test/circt/build/bin/./firtool --verilog --lowering-options=exprInEventControl,verifLabels --disable-annotation-unknown \
  --annotation-file=/home/lsteveol/projects/wlink_test/wav-wlink-hw/wav.wlink.AXI128bit6LaneWlinkTestConfig/WlinkSimpleTestHarness.anno.json \
  /home/lsteveol/projects/wlink_test/wav-wlink-hw/wav.wlink.AXI128bit6LaneWlinkTestConfig/WlinkSimpleTestHarness.fir -o myverilog.v
```

Does FIRRTL with Annotations require some sanitation prior to `firtool`? I've been trying to find some docs, but haven't had much luck.

seldridge, 2021-11-29:
Any "normal" annotation (one that extends `SingleTargetAnnotation`) should get applied correctly to the operation in the circuit indicated by the target. That error you're seeing is indicating that the annotation doesn't match to anything in the circuit. I'd first verify that the target, `~WlinkSimpleTestHarness|WlinkGPIOPHY>scan_out` does exist in the input FIRRTL file (if the circuit doesn't match you will get a different error, but you can get this error if either `WlinkGPIOPHY` isn't a module in the circuit or if `scan_out` isn't a wire/port/reg/node inside `WlinkGPIOPHY`.). If it does exist, then there may be an issue on the CIRCT side.

As an example, you can get this with the following:
```scala
circuit Foo :
  module Foo :
    output a: UInt<1>

    a is invalid
```

```json
{
  "class":"hello",
  "target":"~Foo|Foo>x"
}
```

Running `firtool Foo.fir -parse-only -annotation-file Foo.anno.json` produces the following because `x` doesn't exist:
```
Foo.fir:1:1: error: unapplied annotations with target '~Foo|Foo>x' and payload '[{class = "hello"}]'

circuit Foo : %[[
^
```

If your annotations are not consumed by the FIRRTL transforms inside CIRCT, these will be silently dropped unless you run with `-warn-on-unprocessed-annotations`. BoringUtils annotations (`SourceAnnotation`/`SinkAnnotation`) are not yet supported and will produce this warning.

lsteveol, 2021-11-29:
So what I have is a Chisel `Bundle`, and there is a `dontTouch` on this particular signal. The FIRRTL looks  like the following

```
//Some things removed for clarity
  module WlinkGPIOPHY : 
    input clock : Clock
    input reset : AsyncReset
//scan.out (at the endof line)
    output scan : {flip mode : UInt<1>, flip asyncrst_ctrl : UInt<1>, flip clk : UInt<1>, flip shift : UInt<1>, flip in : UInt<1>, out : UInt<1>}

//removed for clairty
    scan is invalid 

//...

    gpio.io.scan.mode <= scan.mode @[Bundles.scala 19:19]
    gpio.io.scan.asyncrst_ctrl <= scan.asyncrst_ctrl @[Bundles.scala 20:19]
    gpio.io.scan.clk <= scan.clk @[Bundles.scala 21:19]
    gpio.io.scan.in <= scan.in @[Bundles.scala 22:19]
    gpio.io.scan.shift <= scan.shift @[Bundles.scala 23:19]
```
So there is a `scan_out` signal, however I'm not driving it. So it appears like this is there. I don't know if the code is being eliminated/optimized prior to the annotation trying to apply however.

This is built on Chisel 3.4.

seldridge, 2021-11-29:
That input file looks like High FIRRTL/CHIRRTL (because the bundles are not lowered), but the annotation file has flattened bundles. In that circuit, `~Blah|WlinkGPIOPHY>scan_out` technically does not exist, but `~Black|WlinkGPIOPHY>scan.out` _does exist_. It looks like you may be using an SFC-produced annotation file with CHIRRTL input?

The SFC may overwrite the annotation file if you do something like `ChiselStage.emitVerilog(...)`. You can try using `ChiselStage.emitChirrtl(...)` and using that FIRRTL file / annotation file.

seldridge, 2021-11-30:
The `-verilog` output is expected (and useful for our own testing as this enables us to write concise tests that look for specific files). However, this is clearly not legal Verilog that we're outputting. I'd open an issue about fixing this. At minimum, running with `-verilog` should not produce a `.f` file if it isn't necessary since everything is in one file.

For the time being, is `-split-verilog` sufficient for your use case?

lsteveol, 2021-11-30:
Ok thanks! I'll open another ticket.

`-split-verilog` is good for me at the moment. I'm mainly starting to checkout the differences between the FIRRTL verilog emitter and CIRCT, so nothing is holding me up.

closed 2021-11-30 by lsteveol

seldridge, 2021-12-01:
Nice.  Thanks for checking out CIRCT, @lsteveol!

lsteveol, 2021-12-01:
I opened https://github.com/llvm/circt/issues/2266. Thanks @seldridge !