llvm/circt #2255: [FIRRTL] APInt and Zero/Sign Extension Pain

seldridge, 2021-11-30:
While `APInt` has been generally extended to handle zero-width things in upstream and this has been turned on in02a4839129c2b7dfecf2205b23304bf4177166a0, many of the methods around zero/sign extension result in assertions firing.  I've noticed the following:

1. At least one assertion inside `llvm/ADT/APInt.h` disallows usage of zero-width APInts for sign extension when it probably shouldn't.
2. Sign extension falls back to methods inside `llvm/Support/MathExtras.h` which assert on zero-width.

An example of such a failure showing (1) is:

```mlir
firrtl.circuit "top_mod"   {
  firrtl.module @top_mod(in %a: !firrtl.uint<1>, out %b: !firrtl.uint<1>) {
    %zero = firrtl.constant 0 : !firrtl.uint<0>
    %0 = firrtl.dshl %a, %zero : (!firrtl.uint<1>, !firrtl.uint<0>) -> !firrtl.uint<1>
    firrtl.connect %b, %0 : !firrtl.uint<1>, !firrtl.uint<1>
  }
}
```

Both (1) and (2) are the source of nightly performance regressions crashing: https://github.com/circt/perf/blob/gh-pages/test1-2021-11-29.log

We can kludge around (1) with a local fix. I can't se a way around (2) without upstream LLVM changes either with some simple fixes to remove assertions or by defining more zero-width APInt behavior that doesn't use MathExtras. I have a candidate LLVM fix that takes the former, simpler approach: https://reviews.llvm.org/D114768

<details>
  <summary>Example kludge for (1).</summary>
  
```diff
diff --git a/lib/Dialect/FIRRTL/FIRRTLFolds.cpp b/lib/Dialect/FIRRTL/FIRRTLFolds.cpp
index efef5018..f6199d7a 100644
--- a/lib/Dialect/FIRRTL/FIRRTLFolds.cpp
+++ b/lib/Dialect/FIRRTL/FIRRTLFolds.cpp
@@ -33,6 +33,13 @@ namespace patterns {
 using namespace circt;
 using namespace firrtl;
 
+/// Temporary fix until getLimitedValue is defined for zero-width APInt.
+static uint64_t getLimitedValueKludge(APInt a, uint64_t limit = UINT64_MAX) {
+  if (a.getBitWidth() == 0)
+    return 0;
+  return a.getLimitedValue(limit);
+}
+
 static IntegerAttr getIntAttr(Type type, const APInt &value) {
   auto intType = type.cast<IntType>();
   assert((!intType.hasWidth() ||
@@ -104,8 +111,14 @@ static Optional<APSInt> getExtendedConstant(Value operand, Attribute constant,
     return {};
 
   // Extension signedness follows the operand sign.
-  if (IntegerAttr result = constant.dyn_cast_or_null<IntegerAttr>())
+  if (IntegerAttr result = constant.dyn_cast_or_null<IntegerAttr>()) {
+    auto value = result.getAPSInt();
+    // TODO: This is a kludge to get around extOrTrunce throwing an assert on
+    // zero-width values.
+    if (!value.getBitWidth())
+      return APSInt(destWidth, value.isUnsigned());
     return result.getAPSInt().extOrTrunc(destWidth);
+  }
 
   // If the operand is zero bits, then we can return a zero of the result
   // type.
@@ -938,7 +951,7 @@ LogicalResult DShlPrimOp::canonicalize(DShlPrimOp op,
   // what is needed for the constant.
   if (auto rhsCst = dyn_cast_or_null<ConstantOp>(op.rhs().getDefiningOp())) {
     // Shift amounts are always unsigned, but shift only takes a 32-bit amount.
-    uint64_t shiftAmt = rhsCst.value().getLimitedValue(1ULL << 31);
+    uint64_t shiftAmt = getLimitedValueKludge(rhsCst.value(), 1ULL << 31);
     auto result =
         rewriter.createOrFold<ShlPrimOp>(op.getLoc(), op.lhs(), shiftAmt);
     rewriter.replaceOpWithNewOp<PadPrimOp>(
@@ -957,7 +970,7 @@ LogicalResult DShrPrimOp::canonicalize(DShrPrimOp op,
   // what is needed for the constant.
   if (auto rhsCst = dyn_cast_or_null<ConstantOp>(op.rhs().getDefiningOp())) {
     // Shift amounts are always unsigned, but shift only takes a 32-bit amount.
-    uint64_t shiftAmt = rhsCst.value().getLimitedValue(1ULL << 31);
+    uint64_t shiftAmt = getLimitedValueKludge(rhsCst.value(), 1ULL << 31);
     auto result =
         rewriter.createOrFold<ShrPrimOp>(op.getLoc(), op.lhs(), shiftAmt);
     rewriter.replaceOpWithNewOp<PadPrimOp>(
```

</details>

lattner, 2021-11-30:
Yes, we should work around this until and if APInt is fixed.

For rationale on why APInt is defined this way, please see the discussion [here](https://reviews.llvm.org/D109555) and [here](https://reviews.llvm.org/D111241).

I think that defining sext/zext from zero bit as zero is defensible!

seldridge, 2021-12-01:
Nightly performance regressions are back to passing after 97da1889183d830061b306c0dbf713da1f3be76b and 6479b4e712666681dd54455b2bbf3a12e5e320f8.  There are now some simple support functions added in 033486f0c6a0eccc33e91f4688011c03ec8ca345 for working with APInt/APSInt and getting around the assertions in upstream MLIR.

closed 2021-12-01 by seldridge