llvm/circt #2295: [FIRRTL/IMConstantProp] Explosive time consumption of IMCP

uenoku, 2021-12-06:
From https://github.com/llvm/circt/commit/c351dc601722fbb3d5884ce786bf67321c63b0dc, we are spending explosive time(50 minutes!) on IMCP  for the chipyard test.
[image]

I suspect this comes from the visiting order of operations in ICMP. Currently, ICMP visits an operation immediately if a lattice value of one of its operands changes. I think this is the problem: Fox example,
```
registers: reg1, reg2, reg3, reg4, ...., reg_m

reg1 <= UInt(1)
reg2 <= UInt(2)
reg3 <= UInt(3)
....
reg_m <= UInt(m)

sum <= add(reg1, reg2, reg3, ..., reg_m)

use1 <= op(sum)
use2 <= op(use1)
use3 <= op(use2)
use4 <= op(use3)
...
use_n <= op(use_{n-1}) 
```
In this example, we first visit reg1's initialization and then we visit all use-chains of `reg1`. So we visit `sum, use1, use2, use3, ..., use_n` (under the condition that `reg2, reg3 ...., reg_m` are invalid values). After that, we visit `reg2`'s initialization and then we again compute all the values of `sum, use1, use2, ..., use_n`  so it will become O(NM) computations eventually. If we change the traversal order from LIFO to FIFO we can initialize all the registers before visiting their use-chains.

This behaviour is not exposed because we gave up invalid canonicalizations in many cases before.

uenoku, 2021-12-06:
I opened the fix PR for now https://github.com/llvm/circt/pull/2297

uenoku, 2021-12-08:
[image]

Ok, it looks fine

lattner, 2021-12-08:
Thank you for catching this, should this issue be closed now?

uenoku, 2021-12-08:
Yes, I'm fine with closing it. We might want to use SCC based approach in the end but the current approach seems sufficient for now.

closed 2021-12-08 by uenoku