llvm/circt #2298: [LowerToHW] Define Behavior for Multiple Circuits

seldridge, 2021-12-06:
Currently, if you compile multiple FIRRTL circuits, only one of them gets lowered. We need to define a behavior here for how this should work and what the resulting HW representation is. Consider the following:

```mlir
module  {
  firrtl.circuit "Foo"   {
    firrtl.module @Foo(in %a: !firrtl.uint<1>, out %b: !firrtl.uint<1>) {
      firrtl.connect %b, %a : !firrtl.uint<1>, !firrtl.uint<1>
    }
  }
  firrtl.circuit "Bar"   {
    firrtl.module @Bar(in %a: !firrtl.uint<1>, out %b: !firrtl.uint<1>) {
      firrtl.connect %b, %a : !firrtl.uint<1>, !firrtl.uint<1>
    }
  }
}
```

This produces, after `LowerToHW`:
```mlir
module  {
  firrtl.circuit "Bar"   {
    firrtl.module @Bar(in %a: !firrtl.uint<1>, out %b: !firrtl.uint<1>) {
      firrtl.connect %b, %a : !firrtl.uint<1>, !firrtl.uint<1>
    }
  }
  hw.module @Foo(%a: i1) -> (b: i1) {
    hw.output %a : i1
  }
}
```

If allowed to continue after this, eventually `ExportVerilog` will print an error because it doesn't know how to handle a `firrtl.circuit`.

I can think of two behaviors here:

1. This should produce a flat list of modules.
2. (Likely what we need) This should lower each to two lists of modules in a `hw.namespace`-guarded region.

lattner, 2021-12-08:
I think that we should just reject this until there is a need for it (link time optimization??).  There are probably a few unit tests incorrectly using this, but they can switch to split files form

lattner, 2021-12-08:
Even better would be to get rid of the top level mlir::ModuleOp entirely for FIRRTL dialect, and add a new hw.design top level op.

seldridge, 2021-12-08:
> I think that we should just reject this until there is a need for it (link time optimization??).

The current use case, which @albertchen-sifive can comment more on, is that two Grand Central features need to ingest a `SubCircuitAnnotation` with members:
- a FIRRTL Circuit
- some annotations
- a package name

We're looking for some IR representation for this and the original idea of "just using multiple circuits" ran into the issue here.