llvm/circt #2299: [FIRRTL] Parsing a `-1` Parameter Results in SFC Divergence

seldridge, 2021-12-06:
When parsing external module parameters from FIRRTL text, negative numbers need to be expanded to a 32-bit value.  (Likely, all integer parameters should just be parsed as 32-bit values regardless of whether or not they can fit in a smaller-sized APSInt.)

As an example, the following input FIRRTL text:

```scala
circuit Foo:
  extmodule Bar:
    output a: UInt<1>
    parameter x = -1
  module Foo:
    output a: UInt<1>

    inst bar of Bar
    a <= bar.a
```

Is parsed into:
```mlir
firrtl.circuit "Foo"   {
  firrtl.extmodule @Bar(out a: !firrtl.uint<1>) attributes {parameters = {x = -1 : i4}}
  firrtl.module @Foo(out %a: !firrtl.uint<1>) {
    %bar_a = firrtl.instance bar  @Bar(out a: !firrtl.uint<1>)
    firrtl.connect %a, %bar_a : !firrtl.uint<1>, !firrtl.uint<1>
  }
}
```

This results in the following Verilog:
```verilog
// external module Bar

module Foo(     // Foo.fir:5:10
  output a);

  Bar #(
    .x(15)
  ) bar (       // Foo.fir:8:5
    .a (a)
  );
endmodule
```

The SFC produces:
```verilog
module Foo(
  output  a
);
  wire  bar_a;
  Bar #(.x(-1)) bar (
    .a(bar_a)
  );
  assign a = bar_a;
endmodule
```

fabianschuiki, 2021-12-06:
Might this just be an issue of honoring the sign of the parameter upon emission (at least until we add proper parameters with types)?

seldridge, 2021-12-06:
> Might this just be an issue of honoring the sign of the parameter upon emission (at least until we add proper parameters with types)?

Possibly. Though, I can see a situation where a parameter that is truly a `-1 : i4` should print as `15` if the policy is to print Verilog parameters as unsigned. I tried to reflect this in the end-to-end test that #2300 adds to indicate that either `-1` or `4294967295` should be fine, just `15` is not.

PR #2300 (2021-12-06): [FIRRTL] Parse params < 32-bit as 32-bit
Change FIRRTL parsing to convert all parameters less than 32-bit to
32-bit APInts.  This fixes downstream issues with Verilog emission,
where negative parameters may have too few bits in their APInt
representation and print as shorter-than-expected literals.
    
Fixes #2299.

This follows the minimal suggestion of [this comment](https://github.com/llvm/circt/pull/2076#issuecomment-962184122) to just store parameters as 32-bit values.
commit e5f0c506 (2021-12-06): [FIRRTL] Parse params < 32-bit as 32-bit (#2300)

Change FIRRTL parsing to convert all parameters less than 32-bit to
32-bit APInts.  This fixes downstream issues with Verilog emission,
where negative parameters may have too few bits in their APInt
representation and print as shorter-than-expected literals.

Fixes #2299.
closed 2021-12-06 by seldridge