llvm/circt #2310: [SV] Randomization Sharing Leads to Xs with disallowLocalVariables

seldridge, 2021-12-08:
If `-lowering-options=disallowLocalVariables` is used to generate Verilog, you can create situations where registers are not randomized in the evaluation of the initial block.

Consider the following source FIRRTL:

```firrtl
circuit Foo:
  module Foo:
    input clock: Clock
    input d: UInt<16>[2]
    output q: UInt<16>[2]

    reg r: UInt<16>, clock
    reg s: UInt<16>, clock

    r <= d[0]
    q[0] <= r

    s <= d[1]
    q[1] <= s
```

This has two registers of size 16 which will result in sharing of the `RANDOM` which is size 32.  This will eventually produce the following Verilog using `firtool -verilog -lowering-options=disallowLocalVariables`:

```verilog
module Foo(	// Foo.fir:2:10
  input         clock,
  input  [15:0] d_0, d_1,
  output [15:0] q_0, q_1);

  reg [15:0] r;	// Foo.fir:7:5
  reg [15:0] s;	// Foo.fir:8:5

  `ifndef SYNTHESIS	// Foo.fir:7:5
    reg [31:0] _T;	// Foo.fir:7:5

  wire [15:0] _T_15to0 = _T[15:0];	// Foo.fir:7:5
  wire [15:0] _T_31to16 = _T[31:16];	// Foo.fir:7:5, :8:5
    initial begin	// Foo.fir:7:5
      `INIT_RANDOM_PROLOG_	// Foo.fir:7:5
      `ifdef RANDOMIZE_REG_INIT	// Foo.fir:7:5
        _T = `RANDOM;	// Foo.fir:7:5
        r = _T_15to0;	// Foo.fir:7:5
        s = _T_31to16;	// Foo.fir:8:5
      `endif
    end // initial
  `endif
  always @(posedge clock) begin	// Foo.fir:10:7
    r <= d_0;	// Foo.fir:10:7
    s <= d_1;	// Foo.fir:13:7
  end // always @(posedge)
  assign q_0 = r;	// Foo.fir:2:10
  assign q_1 = s;	// Foo.fir:2:10
endmodule
```

The spilling of the wire out of the procedural initial region is wrong as far as I can tell.  What happens is this:

1. `_T` gets a random value
2. `r` gets the value of `_T_15to0` (which is currently `x`) and has not been updated to `_T[15:0]`
3. `s` also gets an `x` for the same reason as (2)
4. `_T_15to0` and `_T_31to16` get updated

This is tedious because we really can't spill outside of this procedural region without changing semantics.

seldridge, 2021-12-08:
I think the only way to fix this is to spill the variable declaration (which can just be a `reg`) and then the expression stays in the procedural region:

Something like:
```verilog
reg _T_0;
initial begin 
  _T = `RANDOM;
  _T_0 = _T[0];
  r = _T_0;
end
```

lattner, 2021-12-09:
also really odd the _T_0 wire isn't indented right.

lattner, 2021-12-13:
I've been pondering on this, and I don't think there is a really great answer.  If this is a blocker, I'd recommend a hack to just disable `hoistNonSideEffectExpr` when inside an initial region.  That should solve this problem, but will cause overly long expressions in initial blocks to generate 'automatic' declarations (and assert out).  If things seem to work, then it is probably possible to limp on that for quite a while.

The "right" answer is to eliminate "hoistNonSideEffectExpr" entirely: it isn't great for readability and code quality to proactively hoist all expressions, because it generates a ton of extraneous wires, even for silly cases like this.  The better answer is for ExportVerilog to lazily "spill" to wires only expressions that need it, just like we spill to "automatic logic" when needed.  This will be a significantly more complicated change, but is the right way to go.

lattner, 2021-12-13:
> also really odd the _T_0 wire isn't indented right.

This tiny sub-bug is fixed in d4586238

PR #2380 (2021-12-23): [LowerToHW] Fix reg init via verbatim ops
Work around a bug in Verilog emission with the disallowLocalVariables
option where temporaries are spilled to wires which violates the
ordering of the initial block used for register randomization.  This
manifests as X's showing up in registers after initialization.  This
avoids this entirely by using explicit registers and verbatim operations
using those registers.

Fixes #2310.

For the example FIRRTL circuit shown in #2310:

```verilog
circuit Foo:
  module Foo:
    input clock: Clock
    input d: UInt<1>
    output q: UInt<1>

    reg r: UInt<1>, c [... cut]
closed 2021-12-23 by seldridge