llvm/circt #2342: [ExportVerilog] Invalid emission of mux of aggregates + extract

uenoku, 2021-12-14:
```mlir
// circt-opt %s -export-verilog
hw.module @muxAggregate(%cond: i1, %a: !hw.struct<c: i1>, %b: !hw.struct<c: i1>) -> (out: i1) {
    %0 = comb.mux %cond, %a, %b : !hw.struct<c: i1>
    %1 = hw.struct_extract %0["c"] : !hw.struct<c: i1>
    hw.output %1 : i1
}
```
will become
```sv
module muxAggregate(      // foo.mlir:2:3
  input                            cond,
  input  struct packed {logic c; } a, b,
  output                           out);

  assign out = (cond ? a : b).c;        // foo.mlir:3:12, :4:12, :5:5
endmodule
```
`(cond?a:b).c` causes a syntax error so we need to spil mux into a temporary wire.

lattner, 2021-12-15:
I'm not exactly sure what the rules are in Verilog, but the fix should be in `isExpressionUnableToInline` in ExportVerilog.cpp

uenoku, 2021-12-15:
Thanks! It was fixed by just adding operations to `isExpressionUnableToInline`.

closed 2021-12-15 by uenoku