llvm/circt #2349: Failure to compile CIRCT with provided README instructions: `...internal compiler error: in replace_placeholders_r...`

mkorbel1, 2021-12-15:
I'm trying to build CIRCT following the instructions on the README, but I'm hitting a compile error.  I was able to build LLVM in release mode (debug mode ran out of memory on my laptop), but the build for CIRCT fails with the below error (`...internal compiler error: in replace_placeholders_r...`).  I've tried various minor tweaks but none of them have been successful at bypassing the error.

I'm building using WSL2 with Ubuntu on Windows 10, if that is relevant.

```
$ cmake -G Ninja .. \
>     -DMLIR_DIR=$PWD/../llvm/build/lib/cmake/mlir \
>     -DLLVM_DIR=$PWD/../llvm/build/lib/cmake/llvm \
>     -DLLVM_ENABLE_ASSERTIONS=ON \
>     -DCMAKE_BUILD_TYPE=DEBUG \
>     -DCMAKE_EXPORT_COMPILE_COMMANDS=ON
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11")
-- Using MLIRConfig.cmake in: /home/max/work/circt/build/../llvm/build/lib/cmake/mlir
-- Using LLVMConfig.cmake in: /home/max/work/circt/build/../llvm/build/lib/cmake/llvm
-- Linker detection: GNU ld
-- Performing Test C_SUPPORTS_FPIC
-- Performing Test C_SUPPORTS_FPIC - Success
-- Performing Test CXX_SUPPORTS_FPIC
-- Performing Test CXX_SUPPORTS_FPIC - Success
-- Building with -fPIC
-- Performing Test C_SUPPORTS_FNO_SEMANTIC_INTERPOSITION
-- Performing Test C_SUPPORTS_FNO_SEMANTIC_INTERPOSITION - Success
-- Performing Test CXX_SUPPORTS_FNO_SEMANTIC_INTERPOSITION
-- Performing Test CXX_SUPPORTS_FNO_SEMANTIC_INTERPOSITION - Success
-- Performing Test SUPPORTS_FVISIBILITY_INLINES_HIDDEN_FLAG
-- Performing Test SUPPORTS_FVISIBILITY_INLINES_HIDDEN_FLAG - Success
-- Performing Test C_SUPPORTS_WERROR_DATE_TIME
-- Performing Test C_SUPPORTS_WERROR_DATE_TIME - Success
-- Performing Test CXX_SUPPORTS_WERROR_DATE_TIME
-- Performing Test CXX_SUPPORTS_WERROR_DATE_TIME - Success
-- Performing Test C_SUPPORTS_WERROR_UNGUARDED_AVAILABILITY_NEW
-- Performing Test C_SUPPORTS_WERROR_UNGUARDED_AVAILABILITY_NEW - Failed
-- Performing Test CXX_SUPPORTS_WERROR_UNGUARDED_AVAILABILITY_NEW
-- Performing Test CXX_SUPPORTS_WERROR_UNGUARDED_AVAILABILITY_NEW - Failed
-- Performing Test CXX_SUPPORTS_MISSING_FIELD_INITIALIZERS_FLAG
-- Performing Test CXX_SUPPORTS_MISSING_FIELD_INITIALIZERS_FLAG - Success
-- Performing Test C_SUPPORTS_IMPLICIT_FALLTHROUGH_FLAG
-- Performing Test C_SUPPORTS_IMPLICIT_FALLTHROUGH_FLAG - Success
-- Performing Test CXX_SUPPORTS_IMPLICIT_FALLTHROUGH_FLAG
-- Performing Test CXX_SUPPORTS_IMPLICIT_FALLTHROUGH_FLAG - Success
-- Performing Test C_SUPPORTS_COVERED_SWITCH_DEFAULT_FLAG
-- Performing Test C_SUPPORTS_COVERED_SWITCH_DEFAULT_FLAG - Failed
-- Performing Test CXX_SUPPORTS_COVERED_SWITCH_DEFAULT_FLAG
-- Performing Test CXX_SUPPORTS_COVERED_SWITCH_DEFAULT_FLAG - Failed
-- Performing Test CXX_SUPPORTS_CLASS_MEMACCESS_FLAG
-- Performing Test CXX_SUPPORTS_CLASS_MEMACCESS_FLAG - Failed
-- Performing Test CXX_SUPPORTS_REDUNDANT_MOVE_FLAG
-- Performing Test CXX_SUPPORTS_REDUNDANT_MOVE_FLAG - Failed
-- Performing Test CXX_SUPPORTS_PESSIMIZING_MOVE_FLAG
-- Performing Test CXX_SUPPORTS_PESSIMIZING_MOVE_FLAG - Failed
-- Performing Test CXX_SUPPORTS_NOEXCEPT_TYPE_FLAG
-- Performing Test CXX_SUPPORTS_NOEXCEPT_TYPE_FLAG - Success
-- Performing Test CXX_WONT_WARN_ON_FINAL_NONVIRTUALDTOR
-- Performing Test CXX_WONT_WARN_ON_FINAL_NONVIRTUALDTOR - Failed
-- Performing Test CXX_SUPPORTS_SUGGEST_OVERRIDE_FLAG
-- Performing Test CXX_SUPPORTS_SUGGEST_OVERRIDE_FLAG - Success
-- Performing Test CXX_WSUGGEST_OVERRIDE_ALLOWS_ONLY_FINAL
-- Performing Test CXX_WSUGGEST_OVERRIDE_ALLOWS_ONLY_FINAL - Failed
-- Performing Test C_WCOMMENT_ALLOWS_LINE_WRAP
-- Performing Test C_WCOMMENT_ALLOWS_LINE_WRAP - Failed
-- Performing Test C_SUPPORTS_STRING_CONVERSION_FLAG
-- Performing Test C_SUPPORTS_STRING_CONVERSION_FLAG - Failed
-- Performing Test CXX_SUPPORTS_STRING_CONVERSION_FLAG
-- Performing Test CXX_SUPPORTS_STRING_CONVERSION_FLAG - Failed
-- Performing Test C_SUPPORTS_MISLEADING_INDENTATION_FLAG
-- Performing Test C_SUPPORTS_MISLEADING_INDENTATION_FLAG - Success
-- Performing Test CXX_SUPPORTS_MISLEADING_INDENTATION_FLAG
-- Performing Test CXX_SUPPORTS_MISLEADING_INDENTATION_FLAG - Success
-- Performing Test LINKER_SUPPORTS_COLOR_DIAGNOSTICS
-- Performing Test LINKER_SUPPORTS_COLOR_DIAGNOSTICS - Failed
-- Looking for os_signpost_interval_begin
-- Looking for os_signpost_interval_begin - not found
-- Did not find Verilator.
-- Did not find vivado.
-- Did not find Quartus.
-- Did not find Questa.
-- Did not find yosys.
-- Could NOT find CapnProto (missing: CapnProto_DIR)
-- llhd-sim is enabled.
-- CIRCT Python bindings are disabled.
-- PyCDE project is disabled
-- Configuring done
-- Generating done
-- Build files have been written to: /home/max/work/circt/build
$ ninja
[174/353] Building CXX object lib/Dialect/Calyx/Export/CMakeFiles/obj.CIRCTExportCalyx.dir/CalyxEmitter.cpp.o
In file included from /usr/include/c++/7/cassert:44:0,
                 from ../llvm/llvm/include/llvm/Support/CommandLine.h:35,
                 from ../llvm/mlir/include/mlir/Translation.h:15,
                 from ../lib/Dialect/Calyx/Export/CalyxEmitter.cpp:21:
../lib/Dialect/Calyx/Export/CalyxEmitter.cpp: In member function ‘void {anonymous}::Emitter::emitCalyxControl(mlir::Block*)’:
../lib/Dialect/Calyx/Export/CalyxEmitter.cpp:372:44: warning: suggest parentheses around ‘&&’ within ‘||’ [-Wparenthese
]
            parent->hasTrait<ControlLike>() &&
            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
                "This should only be used to emit Calyx Control structures.");
                ~
[214/353] Building CXX object lib/Dialect/FIRRTL/Transform...CMakeFiles/obj.CIRCTFIRRTLTransforms.dir/InferResets.cpp.o
FAILED: lib/Dialect/FIRRTL/Transforms/CMakeFiles/obj.CIRCTFIRRTLTransforms.dir/InferResets.cpp.o
/usr/bin/c++ -DGTEST_HAS_RTTI=0 -D_DEBUG -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I../llvm/llvm/include -I../llvm/build/include -I../llvm/mlir/include -I../llvm/build/tools/mlir/include -I../include -Iinclude -I../lib/Dialect/FIRRTL/. -fno-exceptions -fno-rtti -fPIC -fno-semantic-interposition -fvisibility-inlines-hidden -Werror=date-time -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wno-missing-field-initializers -Wimplicit-fallthrough -Wno-noexcept-type -Wdelete-non-virtual-dtor -Wno-comment -Wmisleading-indentation -fdiagnostics-color -g  -fno-exceptions -fno-rtti -std=gnu++1z -MD -MT lib/Dialect/FIRRTL/Transforms/CMakeFiles/obj.CIRCTFIRRTLTransforms.dir/InferResets.cpp.o -MF lib/Dialect/FIRRTL/Transforms/CMakeFiles/obj.CIRCTFIRRTLTransforms.dir/InferResets.cpp.o.d -o lib/Dialect/FIRRTL/Transforms/CMakeFiles/obj.CIRCTFIRRTLTransforms.dir/InferResets.cpp.o -c ../lib/Dialect/FIRRTL/Transforms/InferResets.cpp
../lib/Dialect/FIRRTL/Transforms/InferResets.cpp: In member function ‘void {anonymous}::InferResetsPass::implementAsyncReset(mlir::Operation*, circt::firrtl::FModuleOp, mlir::Value)’:
../lib/Dialect/FIRRTL/Transforms/InferResets.cpp:1513:30: internal compiler error: in replace_placeholders_r, at cp/tree.c:2804
              Direction::In}}});
                              ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
[223/353] Building CXX object lib/Dialect/FIRRTL/Transforms/CMakeFiles/obj.CIRCTFIRRTLTransforms.dir/LowerTypes.cpp.o
ninja: build stopped: subcommand failed.
```

youngar, 2021-12-15:
> internal compiler error: in replace_placeholders_r, at cp/tree.c:2804

This is a bug in your version GCC and unrelated to CIRCT.  Can you try a newer compiler?

youngar, 2021-12-15:
I have no idea if this works in WSL but [this](https://github.com/circt/images/blob/trunk/circt-ci-build/Dockerfile#L16) is how we grab the newest LLVM for our docker images:
```
wget https://apt.llvm.org/llvm.sh; \
    chmod +x llvm.sh; \
    ./llvm.sh 13
```

mkorbel1, 2021-12-15:
Thanks, sorry I filed the issue, upgrading gcc from 7 to 11 resolved the error.

closed 2021-12-15 by mkorbel1

youngar, 2021-12-16:
Glad you were able to get things working!