llvm/circt #2355: [FIRRTL] Assertion parsing causing segfault

youngar, 2021-12-16:
When handling verification parsing, `foldWhenEncodedVerifOp` erases a `WhenOp` and then the `LazyLocation` thing tries to set a location on the deleted When op.

```
circuit X:
  module X:
    input p: UInt<1>
    input enable: UInt<1>
    input clock: Clock

    when p:
      printf(clock, enable, "foo [verif-library-assume]<extraction-summary>{\"predicateModifier\":{\"type\":\"noMod\"},\"baseMsg\":\"Assumption does not hold (verification library): \"}</extraction-summary> bar")
      stop(clock, enable, 1)
```
`valgrind -- /scratch/aryoung/circt/build-debug/bin/firtool test.fir` gives the following error message.  The first set of warnings produced by Valgrind which mention `llvm::Twine::isNullary()` should be fixed by #2352.
```
==2239652== Invalid write of size 8
==2239652==    at 0x5BF31C: mlir::Operation::setLoc(mlir::Location) (Operation.h:109)
==2239652==    by 0x5AADA5: (anonymous namespace)::LazyLocationListener::endStatement((anonymous namespace)::FIRParser&) (FIRParser.cpp:1406)
==2239652==    by 0x5B16F5: (anonymous namespace)::FIRStmtParser::parseSimpleStmt(unsigned int) (FIRParser.cpp:2117)
==2239652==    by 0x5B166A: (anonymous namespace)::FIRStmtParser::parseSimpleStmtBlock(unsigned int) (FIRParser.cpp:2109)
==2239652==    by 0x5BAA0F: (anonymous namespace)::FIRCircuitParser::parseModuleBody((anonymous namespace)::FIRCircuitParser::DeferredModuleToParse&) (FIRParser.cpp:3490)
==2239652==    by 0x5BAAC0: (anonymous namespace)::FIRCircuitParser::parseCircuit(llvm::SmallVectorImpl<llvm::MemoryBuffer const*>&, llvm::SmallVectorImpl<llvm::MemoryBuffer const*>&)::{lambda(unsigned long)#1}::operator()(unsigned long) const (FIRParser.cpp:3638)
==2239652==    by 0x5BDFE1: mlir::LogicalResult mlir::failableParallelForEach<llvm::detail::SafeIntIterator<unsigned long, false>, (anonymous namespace)::FIRCircuitParser::parseCircuit(llvm::SmallVectorImpl<llvm::MemoryBuffer const*>&, llvm::SmallVectorImpl<llvm::MemoryBuffer const*>&)::{lambda(unsigned long)#1}>(mlir::MLIRContext*, llvm::detail::SafeIntIterator<unsigned long, false>, mlir::MLIRContext*, (anonymous namespace)::FIRCircuitParser::parseCircuit(llvm::SmallVectorImpl<llvm::MemoryBuffer const*>&, llvm::SmallVectorImpl<llvm::MemoryBuffer const*>&)::{lambda(unsigned long)#1}&&) (Threading.h:49)
==2239652==    by 0x5BD735: mlir::LogicalResult mlir::failableParallelForEach<llvm::iota_range<unsigned long>, (anonymous namespace)::FIRCircuitParser::parseCircuit(llvm::SmallVectorImpl<llvm::MemoryBuffer const*>&, llvm::SmallVectorImpl<llvm::MemoryBuffer const*>&)::{lambda(unsigned long)#1}>(mlir::MLIRContext*, llvm::iota_range<unsigned long>&&, (anonymous namespace)::FIRCircuitParser::parseCircuit(llvm::SmallVectorImpl<llvm::MemoryBuffer const*>&, llvm::SmallVectorImpl<llvm::MemoryBuffer const*>&)::{lambda(unsigned long)#1}&&) (Threading.h:97)
==2239652==    by 0x5BCAF7: mlir::LogicalResult mlir::failableParallelForEachN<(anonymous namespace)::FIRCircuitParser::parseCircuit(llvm::SmallVectorImpl<llvm::MemoryBuffer const*>&, llvm::SmallVectorImpl<llvm::MemoryBuffer const*>&)::{lambda(unsigned long)#1}>(mlir::MLIRContext*, unsigned long, unsigned long, (anonymous namespace)::FIRCircuitParser::parseCircuit(llvm::SmallVectorImpl<llvm::MemoryBuffer const*>&, llvm::SmallVectorImpl<llvm::MemoryBuffer const*>&)::{lambda(unsigned long)#1}&&) (Threading.h:112)
==2239652==    by 0x5BB886: (anonymous namespace)::FIRCircuitParser::parseCircuit(llvm::SmallVectorImpl<llvm::MemoryBuffer const*>&, llvm::SmallVectorImpl<llvm::MemoryBuffer const*>&) (FIRParser.cpp:3641)
==2239652==    by 0x5BBE4C: circt::firrtl::importFIRFile(llvm::SourceMgr&, mlir::MLIRContext*, circt::firrtl::FIRParserOptions) (FIRParser.cpp:3699)
==2239652==    by 0x40B2F4: processBuffer(mlir::MLIRContext&, mlir::TimingScope&, llvm::SourceMgr&, llvm::Optional<std::unique_ptr<llvm::ToolOutputFile, std::default_delete<llvm::ToolOutputFile> > >&) (firtool.cpp:297)
==2239652==  Address 0x63fb9d8 is 24 bytes inside a block of size 160 free'd
==2239652==    at 0x4C3610C: free (vg_replace_malloc.c:538)
==2239652==    by 0x93D274: mlir::Operation::destroy() (Operation.cpp:177)
==2239652==    by 0x93DB23: llvm::ilist_traits<mlir::Operation>::deleteNode(mlir::Operation*) (Operation.cpp:377)
==2239652==    by 0x832088: llvm::iplist_impl<llvm::simple_ilist<mlir::Operation>, llvm::ilist_traits<mlir::Operation> >::erase(llvm::ilist_iterator<llvm::ilist_detail::node_options<mlir::Operation, true, false, void>, false, false>) (ilist.h:268)
==2239652==    by 0x94658F: llvm::iplist_impl<llvm::simple_ilist<mlir::Operation>, llvm::ilist_traits<mlir::Operation> >::erase(mlir::Operation*) (ilist.h:272)
==2239652==    by 0x93DCDA: mlir::Operation::erase() (Operation.cpp:426)
==2239652==    by 0x584C1C: mlir::OpState::erase() (OpDefinition.h:130)
==2239652==    by 0x5EF4ED: circt::firrtl::foldWhenEncodedVerifOp(mlir::ImplicitLocOpBuilder&, circt::firrtl::WhenOp) (FIRParserAsserts.cpp:541)
==2239652==    by 0x5B3C47: (anonymous namespace)::FIRStmtParser::parseWhen(unsigned int) (FIRParser.cpp:2482)
==2239652==    by 0x5B182C: (anonymous namespace)::FIRStmtParser::parseSimpleStmtImpl(unsigned int) (FIRParser.cpp:2162)
==2239652==    by 0x5B16DB: (anonymous namespace)::FIRStmtParser::parseSimpleStmt(unsigned int) (FIRParser.cpp:2116)
==2239652==    by 0x5B166A: (anonymous namespace)::FIRStmtParser::parseSimpleStmtBlock(unsigned int) (FIRParser.cpp:2109)
==2239652==  Block was alloc'd at
==2239652==    at 0x4C34F0B: malloc (vg_replace_malloc.c:307)
==2239652==    by 0x93CB1D: mlir::Operation::create(mlir::Location, mlir::OperationName, mlir::TypeRange, mlir::ValueRange, mlir::DictionaryAttr, mlir::BlockRange, unsigned int) (Operation.cpp:87)
==2239652==    by 0x93C93B: mlir::Operation::create(mlir::Location, mlir::OperationName, mlir::TypeRange, mlir::ValueRange, mlir::DictionaryAttr, mlir::BlockRange, mlir::RegionRange) (Operation.cpp:49)
==2239652==    by 0x93C8C4: mlir::Operation::create(mlir::OperationState const&) (Operation.cpp:38)
==2239652==    by 0x83687B: mlir::OpBuilder::createOperation(mlir::OperationState const&) (Builders.cpp:387)
==2239652==    by 0x5D22D5: circt::firrtl::WhenOp mlir::OpBuilder::create<circt::firrtl::WhenOp, mlir::Value&, bool>(mlir::Location, mlir::Value&, bool&&) (Builders.h:428)
==2239652==    by 0x5C8552: circt::firrtl::WhenOp mlir::ImplicitLocOpBuilder::create<circt::firrtl::WhenOp, mlir::Value&, bool>(mlir::Value&, bool&&) (ImplicitLocOpBuilder.h:67)
==2239652==    by 0x5B3B8D: (anonymous namespace)::FIRStmtParser::parseWhen(unsigned int) (FIRParser.cpp:2438)
==2239652==    by 0x5B182C: (anonymous namespace)::FIRStmtParser::parseSimpleStmtImpl(unsigned int) (FIRParser.cpp:2162)
==2239652==    by 0x5B16DB: (anonymous namespace)::FIRStmtParser::parseSimpleStmt(unsigned int) (FIRParser.cpp:2116)
==2239652==    by 0x5B166A: (anonymous namespace)::FIRStmtParser::parseSimpleStmtBlock(unsigned int) (FIRParser.cpp:2109)
==2239652==    by 0x5BAA0F: (anonymous namespace)::FIRCircuitParser::parseModuleBody((anonymous namespace)::FIRCircuitParser::DeferredModuleToParse&) (FIRParser.cpp:3490)
==2239652==
```

fabianschuiki, 2021-12-16:
I changed the `foldWhenEncodedVerifOp` function in #2357 such that it only removes operations but defers destruction to the `endStatement` call.

uenoku, 2022-01-04:
Fixed by https://github.com/llvm/circt/pull/2388. I confirmed valgrind doesn't give the error message now. So close it.

closed 2022-01-04 by uenoku

youngar, 2022-01-04:
Thanks for following up!