llvm/circt #2363: [ExportVerilog] Weird wire indent level

uenoku, 2021-12-17:
For the following IR, 
```mlir
// ./bin/circt-opt foo.mlir -export-verilog
hw.module @Foo(%a: i1, %b: i1) -> (r1: i1, r2: i1) {
  %0 = comb.icmp eq %a, %b : i1
  hw.output %0, %0 : i1, i1
}
```
ExportVerilog will generate
```sv 
module Foo(     // foo.mlir:2:3
  input  a, b,
  output r1, r2);

wire _T = a == b;       // foo.mlir:3:10
  assign r1 = _T;       // foo.mlir:4:5
  assign r2 = _T;       // foo.mlir:4:5
endmodule
```
The indent level of `_T` should be the same as assignments. Interestingly, the following works correctly. 
```mlir
hw.module @Foo(%a: i1, %b: i1) -> (r1: i1, r2: i1) {
    %0 = comb.icmp eq %a, %b : i1
    %reg = sv.reg  : !hw.inout<i1>
    sv.initial  {
      sv.if %a  {
        sv.passign %reg, %0 : i1
      }
    }
    hw.output %0, %0 : i1, i1
 }
```
```sv
module Foo(     // foo.mlir:2:3
  input  a, b,
  output r1, r2);

  reg reg_0;    // foo.mlir:4:12

  wire _T = a == b;     // foo.mlir:3:10
  initial begin // foo.mlir:5:5
    if (a)      // foo.mlir:6:6
      reg_0 <= a == b;  // foo.mlir:7:13, :8:8
  end // initial
  assign r1 = _T;       // foo.mlir:11:5
  assign r2 = _T;       // foo.mlir:11:5
endmodule
```

I believe this comes from https://github.com/llvm/circt/commit/d45862380c6789f72e8fa92a6c33a188bcd1cf9d. I think `blockDeclarationIndentLevel` is not correctly initialized.

lattner, 2021-12-29:
Agreed, good catch,  I think that blockDeclarationIndentLevel should be initialized to INDENT_AMOUNT.  I'll put together a patch.

lattner, 2021-12-29:
Fixed in bc6666ed

closed 2021-12-29 by lattner