llvm/circt #2372: [FIRRTL] Consider lowering dynamic subaccesses directly to the the

youngar, 2021-12-20:
FIRRTL dynamic subaccess are lowered into a giant tree of mux statements, and then later in the HW canonicalizer, we parse it back into a dynamic array select.  It would be nice if we lowered these subaccess directly to the desired construct without the combinatorial explosion of IR.

```
circuit X:
  module X:
    input index: UInt<4>
    input vec: UInt<8>[16]
    output o: UInt<8>

    o <= vec[index]
```

The parsed IR:
```mlir
firrtl.circuit "X"   {
  firrtl.module @X(in %index: !firrtl.uint<4>, in %vec: !firrtl.vector<uint<8>, 16>, out %o: !firrtl.uint<8>) {
    %0 = firrtl.subaccess %vec[%index] : !firrtl.vector<uint<8>, 16>, !firrtl.uint<4>
    firrtl.connect %o, %0 : !firrtl.uint<8>, !firrtl.uint<8>
  }
}
```

After `firrtl-lower-types` pass:
```mlir
firrtl.module @X(in %index: !firrtl.uint<4>, in %vec_0: !firrtl.uint<8>, in %vec_1: !firrtl.uint<8>, in %vec_2: !firrtl.uint<8>, in %vec_3: !firrtl.uint<8>, in %vec_4: !firrtl.uint<8>, in %vec_5: !firrtl.uint<8>, in %vec_6: !firrtl.uint<8>, in %vec_7: !firrtl.uint<8>, in %vec_8: !firrtl.uint<8>, in %vec_9: !firrtl.uint<8>, in %vec_10: !firrtl.uint<8>, in %vec_11: !firrtl.uint<8>, in %vec_12: !firrtl.uint<8>, in %vec_13: !firrtl.uint<8>, in %vec_14: !firrtl.
uint<8>, in %vec_15: !firrtl.uint<8>, out %o: !firrtl.uint<8>) {
  %c1_ui4 = firrtl.constant 1 : !firrtl.uint<4>
  %0 = firrtl.eq %index, %c1_ui4 : (!firrtl.uint<4>, !firrtl.uint<4>) -> !firrtl.uint<1>
  %1 = firrtl.mux(%0, %vec_1, %vec_0) : (!firrtl.uint<1>, !firrtl.uint<8>, !firrtl.uint<8>) -> !firrtl.uint<8>
  %c2_ui4 = firrtl.constant 2 : !firrtl.uint<4>
  %2 = firrtl.eq %index, %c2_ui4 : (!firrtl.uint<4>, !firrtl.uint<4>) -> !firrtl.uint<1>
  %3 = firrtl.mux(%2, %vec_2, %1) : (!firrtl.uint<1>, !firrtl.uint<8>, !firrtl.uint<8>) -> !firrtl.uint<8>
  %c3_ui4 = firrtl.constant 3 : !firrtl.uint<4>
  %4 = firrtl.eq %index, %c3_ui4 : (!firrtl.uint<4>, !firrtl.uint<4>) -> !firrtl.uint<1>
  %5 = firrtl.mux(%4, %vec_3, %3) : (!firrtl.uint<1>, !firrtl.uint<8>, !firrtl.uint<8>) -> !firrtl.uint<8>
  %c4_ui4 = firrtl.constant 4 : !firrtl.uint<4>
  %6 = firrtl.eq %index, %c4_ui4 : (!firrtl.uint<4>, !firrtl.uint<4>) -> !firrtl.uint<1>
  %7 = firrtl.mux(%6, %vec_4, %5) : (!firrtl.uint<1>, !firrtl.uint<8>, !firrtl.uint<8>) -> !firrtl.uint<8>
  %c5_ui4 = firrtl.constant 5 : !firrtl.uint<4>
  %8 = firrtl.eq %index, %c5_ui4 : (!firrtl.uint<4>, !firrtl.uint<4>) -> !firrtl.uint<1>
  %9 = firrtl.mux(%8, %vec_5, %7) : (!firrtl.uint<1>, !firrtl.uint<8>, !firrtl.uint<8>) -> !firrtl.uint<8>
  %c6_ui4 = firrtl.constant 6 : !firrtl.uint<4>
  %10 = firrtl.eq %index, %c6_ui4 : (!firrtl.uint<4>, !firrtl.uint<4>) -> !firrtl.uint<1>
  %11 = firrtl.mux(%10, %vec_6, %9) : (!firrtl.uint<1>, !firrtl.uint<8>, !firrtl.uint<8>) -> !firrtl.uint<8>
  %c7_ui4 = firrtl.constant 7 : !firrtl.uint<4>
  %12 = firrtl.eq %index, %c7_ui4 : (!firrtl.uint<4>, !firrtl.uint<4>) -> !firrtl.uint<1>
  %13 = firrtl.mux(%12, %vec_7, %11) : (!firrtl.uint<1>, !firrtl.uint<8>, !firrtl.uint<8>) -> !firrtl.uint<8>
  %c8_ui4 = firrtl.constant 8 : !firrtl.uint<4>
  %14 = firrtl.eq %index, %c8_ui4 : (!firrtl.uint<4>, !firrtl.uint<4>) -> !firrtl.uint<1>
  %15 = firrtl.mux(%14, %vec_8, %13) : (!firrtl.uint<1>, !firrtl.uint<8>, !firrtl.uint<8>) -> !firrtl.uint<8>
  %c9_ui4 = firrtl.constant 9 : !firrtl.uint<4>
  %16 = firrtl.eq %index, %c9_ui4 : (!firrtl.uint<4>, !firrtl.uint<4>) -> !firrtl.uint<1>
  %17 = firrtl.mux(%16, %vec_9, %15) : (!firrtl.uint<1>, !firrtl.uint<8>, !firrtl.uint<8>) -> !firrtl.uint<8>
  %c10_ui4 = firrtl.constant 10 : !firrtl.uint<4>
  %18 = firrtl.eq %index, %c10_ui4 : (!firrtl.uint<4>, !firrtl.uint<4>) -> !firrtl.uint<1>
  %19 = firrtl.mux(%18, %vec_10, %17) : (!firrtl.uint<1>, !firrtl.uint<8>, !firrtl.uint<8>) -> !firrtl.uint<8>
  %c11_ui4 = firrtl.constant 11 : !firrtl.uint<4>
  %20 = firrtl.eq %index, %c11_ui4 : (!firrtl.uint<4>, !firrtl.uint<4>) -> !firrtl.uint<1>
  %21 = firrtl.mux(%20, %vec_11, %19) : (!firrtl.uint<1>, !firrtl.uint<8>, !firrtl.uint<8>) -> !firrtl.uint<8>
  %c12_ui4 = firrtl.constant 12 : !firrtl.uint<4>
  %22 = firrtl.eq %index, %c12_ui4 : (!firrtl.uint<4>, !firrtl.uint<4>) -> !firrtl.uint<1>
  %23 = firrtl.mux(%22, %vec_12, %21) : (!firrtl.uint<1>, !firrtl.uint<8>, !firrtl.uint<8>) -> !firrtl.uint<8>
  %c13_ui4 = firrtl.constant 13 : !firrtl.uint<4>
  %24 = firrtl.eq %index, %c13_ui4 : (!firrtl.uint<4>, !firrtl.uint<4>) -> !firrtl.uint<1>
  %25 = firrtl.mux(%24, %vec_13, %23) : (!firrtl.uint<1>, !firrtl.uint<8>, !firrtl.uint<8>) -> !firrtl.uint<8>
  %c14_ui4 = firrtl.constant 14 : !firrtl.uint<4>
  %26 = firrtl.eq %index, %c14_ui4 : (!firrtl.uint<4>, !firrtl.uint<4>) -> !firrtl.uint<1>
  %27 = firrtl.mux(%26, %vec_14, %25) : (!firrtl.uint<1>, !firrtl.uint<8>, !firrtl.uint<8>) -> !firrtl.uint<8>
  %c15_ui4 = firrtl.constant 15 : !firrtl.uint<4>
  %28 = firrtl.eq %index, %c15_ui4 : (!firrtl.uint<4>, !firrtl.uint<4>) -> !firrtl.uint<1>
  %29 = firrtl.mux(%28, %vec_15, %27) : (!firrtl.uint<1>, !firrtl.uint<8>, !firrtl.uint<8>) -> !firrtl.uint<8>
  firrtl.connect %o, %29 : !firrtl.uint<8>, !firrtl.uint<8>
}
```

After the HW `canonicalization` pass:
```mlir
hw.module @X(%index: i4, %vec_0: i8, %vec_1: i8, %vec_2: i8, %vec_3: i8, %vec_4: i8, %vec_5: i8, %vec_6: i8, %vec_7: i8, %vec_8: i8, %vec_9: i8, %vec_10: i8, %vec_11: i8, %vec_12: i8, %vec_13: i8, %vec_14: i8, %vec_15: i8) -> (o: i8) {
  %0 = hw.array_create %vec_15, %vec_14, %vec_13, %vec_12, %vec_11, %vec_10, %vec_9, %vec_8, %vec_7, %vec_6, %vec_5, %vec_4, %vec_3, %vec_2, %vec_1, %vec_0 : i8
  %1 = hw.array_get %0[%index] : !hw.array<16xi8>
  hw.output %1 : i8
}
```

uenoku, 2021-12-20:
This would be duplicate of "adding multi-bit mux" https://github.com/llvm/circt/issues/1617

lattner, 2021-12-20:
Ah, indeed.  Closing since it is a dupe

closed 2021-12-20 by lattner