llvm/circt #2375: [Tracker] Lowering high-level Calyx programs

rachitnigam, 2021-12-21:
Lowering Calyx without having to rely on the native compiler seems to be one of the [top requirements](https://github.com/cucapra/calyx/discussions/804) for folks wanting to use Calyx within CIRCT. This tracks the main compiler passes that need to be implemented to accomplish it.

Calyx has two abstractions that need to be encoded structurally: groups and control operators. The main compilation passes first remove control operators by generating new groups and then removes groups by inlining uses of their `go` and `done` signals.

- [ ] [CompileControl](https://docs.calyxir.org/source/calyx/passes/struct.TopDownCompileControl.html): Removes control operators by encoding them into new groups which implement structural FSMs.
- [ ] [GoInsertion](https://docs.calyxir.org/source/calyx/passes/struct.GoInsertion.html): Guards assignments within a group with a `go` signal that ensures assignments only execute when the group is activated using the FSM.
- [ ] [Inliner](https://docs.calyxir.org/source/calyx/passes/struct.Inliner.html): This pass can be implemented using a simpler, wire-based inliner. Before implementing, ping @rachitnigam to see if this simpler pass has been implemented in the native compiler.

rachitnigam, 2021-12-21:
CC @mikeurbach since you mentioned you are interested in seeing this get done.

cgyurgyik, 2021-12-21:
Current prototypes for these found in [lib/Dialect/Calyx/Transforms](https://github.com/llvm/circt/tree/main/lib/Dialect/Calyx/Transforms). These are (likely) neither correct nor complete, and probably have suffered some bit rot.

mikeurbach, 2021-12-21:
Thanks for opening this, I am interested in helping here.

One thing I've experimented with before was the use of the CIRCT FSM dialect as a lowering target for Calyx. I wonder if such an IR could help simplify these transforms, especially compile control. For starters, it's probably best to follow the original implementations and just massage the IR into a structural form, since that is a well-understood path by now.

It seems @cgyurgyik already has a good start on the three main lowering phases, so perhaps we can refine the task list a bit? For example, we already have one implementation of compile control, but I believe the preferred algorithm is not the one Chris originally implemented, so one task might be to update that and see if there is an improvement. Any other pieces that are incorrect or incomplete? There is also the work to go from structural Calyx to CIRCT's structural IRs, which is currently just a [placeholder](https://github.com/llvm/circt/blob/main/lib/Conversion/CalyxToHW/CalyxToHW.cpp).

As I have time to pick up tasks here and there, I will chime in on this thread.

rachitnigam, 2021-12-22:
Refining the task list sounds good. To start off with, the compile control implementation does indeed use the old, inside-out compilation algorithm that instantiates on FSM for each control operator. The native compiler has switched to using a different pass which compiles the FSM in one shot. Additionally, the current pass only supports [compiling `seq` operators](https://github.com/llvm/circt/blob/main/lib/Dialect/Calyx/Transforms/CompileControl.cpp#L54).

mortbopet, 2022-07-26:
Just to update: I've been pushing on an alternative path to CompileControl, a path going through the FSM dialect (14ca56c139412084fefe62491149f82d77b9b47d, e04b7b6bede7629ef9b067782645d1419263a3bc). Work on this is almost finished (last part is #3601) whereafter control schedule lowering could be considered done.

rachitnigam, 2022-07-26:
Awesome! Looking forward to the PR!

closed 2023-01-31 by mortbopet