llvm/circt #2384: [FIRRTL] FIRParserAsserts fails to detect print-encoded assertions when there is a subelement access

uenoku, 2021-12-23:
```scala
circuit Foo:
  module Foo :
    input cond: UInt<1>
    output io : { clock : Clock}

    when cond :
        printf(io.clock, UInt<1>("h0"), "assert: bar")
```
`circt-translate --import-firrtl` generates
```mlir
firrtl.module @Foo(in %cond: !firrtl.uint<1>, out %io: !firrtl.bundle<clock: clock>) {
    %c0_ui1 = firrtl.constant 0 : !firrtl.uint<1>
    firrtl.when %cond  {
      %0 = firrtl.subfield %io(0) : (!firrtl.bundle<clock: clock>) -> !firrtl.clock
      firrtl.printf %0, %c0_ui1, "assert: bar" 
    }
}
```
The last print op must be converted into firrtl.assert by FIRParserAsserts. The bug is that we only pick when + printf pattern here:
https://github.com/llvm/circt/blob/e5c4eacd1a9550bc51be65e70ed1f77618df2aa9/lib/Dialect/FIRRTL/Import/FIRParserAsserts.cpp#L240-L243

We must accept when + some other ops + printf pattern as well.

fabianschuiki, 2021-12-23:
This is an excellent catch @uenoku! That will require some additional smarts in `FIRParserAsserts.cpp` to pattern match the assert correctly and then properly move the subfield ops to parent blocks. I guess this goes beyond subfield as well, also failing for stuff like `printf(not(io), ...)`.

uenoku, 2021-12-23:
>  I guess this goes beyond subfield as well, also failing for stuff like printf(not(io), ...).

Yes, I agree.

lattner, 2021-12-23:
cool thanks for tracking this fix in the parser!

lattner, 2021-12-23:
One random thought: if it is too painful to handle this "on the fly" in the parser, you could use a little cleanup IR walk that happens as "part of" the parser API, but that gets invoked after each module is parsed.

uenoku, 2021-12-23:
> One random thought: if it is too painful to handle this "on the fly" in the parser, you could use a little cleanup IR walk that happens as "part of" the parser API, but that gets invoked after each module is parsed.

Yeah, that sounds very nice. Probably it's better to move foldWhenEncodedVerifOp logic into post parse processing. 
We can do the same thing in parallel so it's beneficial in terms of parser performance too.

youngar, 2021-12-23:
@uenoku You probably are already aware, but doing this could help with the issue here: https://github.com/llvm/circt/pull/2357

uenoku, 2021-12-24:
> @uenoku You probably are already aware, but doing this could help with the issue here: #2357

Oh, that sounds nice! I didn't notice it. I think deleting operations at parsing is super dangerous so moving the logic to post parsing would be good way to go.

closed 2021-12-27 by uenoku