llvm/circt #2393: [firtool] Run HWCleanup pass again after canonicalizer

uenoku, 2021-12-26:
`./bin/firtool ../perf/regress/chipyard.TestHarness.RocketSmall1Medium1Big1_BoomMedium1Large1Mega1.top.v.hi.fir` now generates something like this (in always block of `TLMonitor` module):
```sv
    automatic logic _T_397 = _T_217 & ~(|a_first_counter);      // Edges.scala:229:28, :230:25, Monitor.scala:396:20

    if (_T_397) // Monitor.scala:397:15
      opcode <= io_in_a_bits_opcode;    // Monitor.scala:397:15
    if (_T_397) // Monitor.scala:398:15
      param <= io_in_a_bits_param;      // Monitor.scala:398:15
    if (_T_397) // Monitor.scala:399:15
      size <= io_in_a_bits_size;        // Monitor.scala:399:15
    if (_T_397) // Monitor.scala:400:15
      source <= io_in_a_bits_source;    // Monitor.scala:400:15
    if (_T_397) // Monitor.scala:401:15
      address <= io_in_a_bits_address;
```
We expect these if statements are merged into one statement.  This is because canonicalizer for `PAssignOp` creates new IfOps (note that  HWCleanup is executed before canonicalization).

https://github.com/llvm/circt/blob/4e20f1ad18eda5b6d03f9f96e5f0fc08a15a12a6/lib/Dialect/SV/SVOps.cpp#L1272-L1274

If we run HWCleanup again after canonicalization, we can merge them into one if statement: 
```sv
if (_T_217 & ~(|a_first_counter)) begin     // Edges.scala:229:28, :230:25, Monitor.scala:396:20, :401:15
      opcode <= io_in_a_bits_opcode;    // Monitor.scala:397:15
      param <= io_in_a_bits_param;      // Monitor.scala:398:15
      size <= io_in_a_bits_size;        // Monitor.scala:399:15
      source <= io_in_a_bits_source;    // Monitor.scala:400:15
      address <= io_in_a_bits_address;  // Monitor.scala:401:15
 end
```

I think HWCleanup is cheaper enough compared to other passes, so I would like to suggest to add HWCleanup pass to the later pipeline.

lattner, 2021-12-26:
Yes, I've been wondering about this but never got a chnace to look at it.  Is it enough to turn firtool.cpp:422 into:

```
      modulePM.addPass(sv::createHWCleanupPass());
      modulePM.addPass(createCSEPass());
      modulePM.addPass(createSimpleCanonicalizerPass());
```

into:

```
      modulePM.addPass(createCSEPass());
      modulePM.addPass(createSimpleCanonicalizerPass());
      modulePM.addPass(sv::createHWCleanupPass());
```

Or do we have to run it twice?  Either approach would be fine, but running it once at the right point would be better.  Thanks for digging into this!

uenoku, 2021-12-26:
I thought HWCleanup is designed to run early in the pipeline to introduce more canonicalizations:  
https://github.com/llvm/circt/blob/853cd4eb3b87db866461fafc1b57e2953a0c99c4/lib/Dialect/SV/Transforms/HWCleanup.cpp#L9-L12

I tested following two pipelines for some large circuits but there is no difference. So I think it's fine to run this only once. Thanks!
```
 modulePM.addPass(createCSEPass());
 modulePM.addPass(createSimpleCanonicalizerPass());
 modulePM.addPass(sv::createHWCleanupPass());
```
```
 modulePM.addPass(sv::createHWCleanupPass());
 modulePM.addPass(createCSEPass());
 modulePM.addPass(createSimpleCanonicalizerPass());
 modulePM.addPass(sv::createHWCleanupPass());
```

lattner, 2021-12-27:
cool!

closed 2021-12-28 by uenoku