llvm/circt #2394: [HWCleanup] Missed optimization for nested regions

uenoku, 2021-12-27:
HWCleanup fails to optimize the following IR:
```mlir
// ./bin/circt-opt -hw-cleanup
hw.module @foo() {
  sv.initial {
    sv.ifdef.procedural "FOO" {
      sv.fwrite "A"
    }
  }
  sv.initial {
    sv.ifdef.procedural "FOO" {
      sv.fwrite "B"
    }
  }
}
```
The current output is:
```mlir
 hw.module @foo() {
    sv.initial  {
      sv.ifdef.procedural "FOO"  {
        sv.fwrite "A"
      }
      sv.ifdef.procedural "FOO"  {
        sv.fwrite "B"
      }
    }
    hw.output
}
``` 
But we want these `sv.idef.procedural` to be merged as well, like:  
```mlir
hw.module @foo() {
    sv.initial  {
      sv.ifdef.procedural "FOO"  {
        sv.fwrite "A"
        sv.fwrite "B"
      }
    }
    hw.output
  }
```
This behavior is because we merge sub regions first:
https://github.com/llvm/circt/blob/853cd4eb3b87db866461fafc1b57e2953a0c99c4/lib/Dialect/SV/Transforms/HWCleanup.cpp#L237-L240 

It can be fixed by changing the order of recursions but it would became O(N^2) time complexity if we do this naively. Therefore, we have to do something smart there.

lattner, 2021-12-27:
Great catch.  I suppose we could do two passes: once before visiting the regions in the ops, then visit the ops, then do once after?

uenoku, 2021-12-28:
> Great catch. I suppose we could do two passes: once before visiting the regions in the ops, then visit the ops, then do once after?

It looks great. I'll test it:)

uenoku, 2021-12-28:
> Great catch. I suppose we could do two passes: once before visiting the regions in the ops, then visit the ops, then do once after?

It seems sufficient to merge the current level operation and visit the ops.

closed 2021-12-29 by uenoku