llvm/circt #2409: [HW/SV] Inner name ref to track field names

uenoku, 2021-12-30:
Struct field names sometimes have invalid verilog names such as "begin", "end", "repeat", "vectored" and etc. 
These invalid field names will be renamed at ExportVerilog with https://github.com/llvm/circt/pull/2359, but 
we may want to refer that name from verbatim ops. For example, GrandCentral creates verbatim ops
to access aggregate if their field names are invalid, we end up emitting invalid verilog.

seldridge, 2021-12-30:
A question is does a field ref point at a _specific name/index_ or is it an _accessor_ to a specific name/index?

Say we have something like: `struct<a: vector<struct<b: i1>, 1>`.

If the accessor is `@foo::@bar::2`, should it be `b` or `a[0].b`?

I'm inclined towards the latter, accessor interpretation as it simplifies transform writing, enables modifications of the type without updates to the format string, and reduces the number of symbols used in a verbatim.

uenoku, 2021-12-31:
> I'm inclined towards the latter, accessor interpretation as it simplifies transform writing, enables modifications of the type without updates to the format string, and reduces the number of symbols used in a verbatim.

Sounds better. Then it might be good time to support field ID in HW dialect.

uenoku, 2022-01-13:
Using probe is a better direction, close it.

closed 2022-01-13 by uenoku