llvm/circt #2420: [FIRRTL] Remove Unused Module Inputs

seldridge, 2022-01-04:
CIRCT is currently _not_ pruning unused non-top module inputs while the Scala FIRRTL Compiler (SFC) is.

Consider the following circuit:

```scala
circuit Foo:
  module Bar:
    input a: UInt<1>
    input b: UInt<1>
    output c: UInt<1>

    c <= a

  module Foo:
    input a: UInt<1>
    input b: UInt<1>
    output c: UInt<1>

    inst bar of Bar

    bar.b <= UInt<1>(1)

    bar.a <= a
    c <= bar.c
```

CIRCT produces (`firtool Foo.fir`):
```verilog
module Bar(	// Foo.fir:2:10
  input  a, b,
  output c);

  assign c = a;	// Foo.fir:2:10
endmodule

module Foo(	// Foo.fir:9:10
  input  a, b,
  output c);

  Bar bar (	// Foo.fir:14:5
    .a (a),
    .b (1'h1),	// Foo.fir:16:14
    .c (c)
  );
endmodule
```

The SFC produces:
```verilog
module Bar(
  input   a,
  output  c
);
  assign c = a;
endmodule
module Foo(
  input   a,
  input   b,
  output  c
);
  wire  bar_a;
  wire  bar_c;
  Bar bar (
    .a(bar_a),
    .c(bar_c)
  );
  assign c = bar_c;
  assign bar_a = a;
endmodule
```

uenoku, 2022-01-04:
Maybe it can be done within IMCP by erasing arguments with Unknown states.

lattner, 2022-01-04:
Handling this in IMCP can make sense since that is mutating the circuit in a massive way and you already know inputs should be dead by virtue of being constant.

That said, I'd recommend a new pass for deleting unused ports.  This will make sure to delete ports that are dead because they aren't used, not just ones that are found to be constants.

closed 2022-01-28 by uenoku